aircrack-ng/rtl8814au #1: Awus 1900 (rtl8814au) injection does not work, usb 3.0 does not work correctly

nidhoggDrag, 2020-10-16:
Hi my friend! The problem is the same as it was on the 8812 driver. The injection does not work. But, I want to note that this driver is very unstable, the sending of packets hangs when the command aireplay-ng, after two packets it hangs forever (ctrl + c), .. And also packets [0 | 0 ACKs], that is, the injection does not work the same as on 8812, but this driver still hangs after the second attempt to send a packet ..

USB 3.0 partially works, .. but, in USB 3.0 mode, it only shows the two networks closest to me, .. does not see the rest of the networks (with usb.2.0 it sees 20+ networks ...) ... some problems with signal strength, apparently ..

Note that the 8812au driver is much more stable, .. yes, .. injection does not work there either (the only drawback!), But at least there are no problems with hanging packages and no problems with usb 3.0 ..

I have not yet met a single fully working Linux driver for my device (awus 1900), .. without bugs, .. it seems that the device is not the cheapest, .. but unfortunately there is no driver for Linux yet ..

awus 1900 8814au + kali

tested both on physical and virtual machines, the result is the same.

root@as:~# git clone -b v5.8.5.1 https://github.com/aircrack-ng/rtl8814au.git
         Клонирование в «rtl8814au»…
         remote: Enumerating objects: 975, done.
         remote: Counting objects: 100% (975/975), done.
         remote: Compressing objects: 100% (613/613), done.
         remote: Total 975 (delta 456), reused 861 (delta 349), pack-reused 0
         Получение объектов: 100% (975/975), 2.94 MiB | 4.89 MiB/s, готово.
         Определение изменений: 100% (456/456), готово.
root@as:~# cd rtl8814au
root@as:~/rtl8814au# make
         /bin/sh: 1: bc: not found
         make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.8.0-kali2-amd64/build M=/root/rtl8814au  modules
         make[1]: вход в каталог «/usr/src/linux-headers-5.8.0-kali2-amd64»
         /bin/sh: 1: bc: not found
  CC [M]  /root/rtl8814au/core/rtw_cmd.o
  CC [M]  /root/rtl8814au/core/rtw_security.o
  CC [M]  /root/rtl8814au/core/rtw_debug.o
  CC [M]  /root/rtl8814au/core/rtw_io.o
  CC [M]  /root/rtl8814au/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8814au/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8814au/core/rtw_ieee80211.o
  CC [M]  /root/rtl8814au/core/rtw_mlme.o
  CC [M]  /root/rtl8814au/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8814au/core/rtw_mi.o
  CC [M]  /root/rtl8814au/core/rtw_wlan_util.o
  CC [M]  /root/rtl8814au/core/rtw_vht.o
  CC [M]  /root/rtl8814au/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8814au/core/rtw_rf.o
  CC [M]  /root/rtl8814au/core/rtw_chplan.o
  CC [M]  /root/rtl8814au/core/rtw_recv.o
  CC [M]  /root/rtl8814au/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8814au/core/rtw_ap.o
  CC [M]  /root/rtl8814au/core/mesh/rtw_mesh.o
  CC [M]  /root/rtl8814au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/rtl8814au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/rtl8814au/core/rtw_xmit.o
  CC [M]  /root/rtl8814au/core/rtw_p2p.o
  CC [M]  /root/rtl8814au/core/rtw_rson.o
  CC [M]  /root/rtl8814au/core/rtw_tdls.o
  CC [M]  /root/rtl8814au/core/rtw_br_ext.o
  CC [M]  /root/rtl8814au/core/rtw_iol.o
  CC [M]  /root/rtl8814au/core/rtw_sreset.o
  CC [M]  /root/rtl8814au/core/rtw_btcoex_wifionly.o
  CC [M]  /root/rtl8814au/core/rtw_btcoex.o
  CC [M]  /root/rtl8814au/core/rtw_beamforming.o
  CC [M]  /root/rtl8814au/core/rtw_odm.o
  CC [M]  /root/rtl8814au/core/rtw_rm.o
  CC [M]  /root/rtl8814au/core/rtw_rm_fsm.o
  CC [M]  /root/rtl8814au/core/rtw_rm_util.o
  CC [M]  /root/rtl8814au/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8814au/os_dep/osdep_service.o
  CC [M]  /root/rtl8814au/os_dep/linux/os_intfs.o
  CC [M]  /root/rtl8814au/os_dep/linux/usb_intf.o
  CC [M]  /root/rtl8814au/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/rtl8814au/os_dep/linux/ioctl_linux.o
/root/rtl8814au/os_dep/linux/ioctl_linux.c:12327:30: warning: ‘rtw_get_wireless_stats’ defined but not used [-Wunused-function]
12327 | static struct iw_statistics *rtw_get_wireless_stats(struct net_device *dev)
      |                              ^~~~~~~~~~~~~~~~~~~~~~
[... 128 lines cut ...]
      addr xx:xx:xx:xx:xx:xx
      type monitor
      wiphy 0
      channel 1 (2412 MHz), width: 40 MHz, center1: 2402 MHz
      txpower 20.00 dBm
root@as:~# 

root@as:~# aireplay-ng --deauth 20 -a xx:xx:xx:xx:xx:xx -c xx:xx:xx:xx:xx:xx wlan0
      00:57:42  Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx:xx) on channel 1
      00:57:42  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
      00:57:43  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
^C
root@as:~# aireplay-ng -0 3 -a xx:xx:xx:xx:xx:xx -c xx:xx:xx:xx:xx:xx wlan0
      01:00:56  Waiting for beacon frame (BSSID: 10:0D:7F:6E:7A:26) on channel 1
      01:00:56  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
      01:00:56  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
^C
root@as:~# aireplay-ng -0 0 -a xx:xx:xx:xx:xx:xx wlan0
      01:01:19  Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx:xx) on channel 1
      NB: this attack is more effective when targeting
      a connected wireless client (-c <client's mac>).
      01:01:19  Sending DeAuth (code 7) to broadcast -- BSSID: [xx:xx:xx:xx:xx:xx]
      01:01:20  Sending DeAuth (code 7) to broadcast -- BSSID: [xx:xx:xx:xx:xx:xx]
      01:01:20  Sending DeAuth (code 7) to broadcast -- BSSID: [xx:xx:xx:xx:xx:xx]
^C

kimocoder, 2020-10-17:
I know, just missing a few more hours to get there. :+1:

hifihedgehog, 2020-10-18:
Thanks so much, @kimocoder ! I just bought a Netgear Nighthawk that has this chip. I noticed it was stuck running in USB 2.0 mode. I am using this driver for openmediavault for serving across my network so I cannot wait for the USB 3.0 speeds!

hifihedgehog, 2020-10-18:
Just as a note to you, @kimocoder , the following needs to be added to CFLAGs in the makefile:

> -Wno-error=date-time

From:
https://askubuntu.com/a/593684

Otherwise, make fails and presents me the following error in Debian Buster on openmediavault:

> error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]

raplin, 2020-10-23:
USB mode switch works for me in current version
`lsusb  -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 25, If 0, Class=Hub, Driver=hub/4p, 5000M
            |__ Port 2: Dev 31, If 0, Class=Vendor Specific Class, Driver=rtl88XXau, 5000M
            |__ Port 4: Dev 33, If 0, Class=Vendor Specific Class, Driver=rtl88XXau, 5000M`
Transfers at superspeed ok (tested on x86 and RPi4)

raplin, 2020-10-23:
USB 3.0 works for me with this driver, in the usual way;
```lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
        |__ Port 1: Dev 25, If 0, Class=Hub, Driver=hub/4p, 5000M
            |__ Port 2: Dev 31, If 0, Class=Vendor Specific Class, Driver=8814au, 5000M
            |__ Port 4: Dev 33, If 0, Class=Vendor Specific Class, Driver=8814au, 5000M
```

kimocoder, 2020-10-23:
USB 3.0 is disabled for now yes, USBModeSwitch needs attention, just like before.. just didn't have time for all.

frame injection capabilities broken when I unlocked all channels and fixing a kernel taint but will be fixed again, when I get more time 👍  well aware

nidhoggDrag, 2020-10-23:
> USB 3.0 is disabled for now yes, USBModeSwitch needs attention, just like before.. just didn't have time for all.
> 
> frame injection capabilities broken when I unlocked all channels and fixing a kernel taint but will be fixed again, when I get more time +1 well aware

my dear friend! I have some news, maybe they will help you in something.

I am not an expert in Kali. But as far as I know, the 88xx drivers are already preinstalled in the Kali kernel. But when installing Kali on a physical machine, my adapter (Awus 1900 rtl8814au) does not work, and I have to manually go to the github and download the drivers and then install them. But why install them if they are already in the kernel ?! But for some reason, those drivers that are embedded in the core of Kali are not active.

Installed Parrot OS recently. And lo and behold! And my adapter immediately worked, and I didn't even have to manually install the driver.

But the most interesting thing is that the drivers that are embedded in the Parrot OS kernel are fully functional! USB 3.0 - it works! The injection works too!

The only cosmetic bug that I found in drivers on Parrot OS is that package information is not displayed:

[0 | 0 ACKs]

BUT! Packages arrive great! Because The device under test disconnects very quickly, if the packets did not reach, the device would not disconnect. (WPA handshake received!)

My guess is that this may be a driver conflict in Kali. You need to somehow activate either the built-in driver in the Kali kernel, or remove it and then install another driver. Two drivers for one device may conflict with each other.

And also I do not quite understand which driver is used in Parrot OS, .. Is it the same driver that is sewn into the Kali kernel, or not?

┌─[root@parrot]─[~]
└──╼ #uname -a
Linux parrot 5.8.0-2parrot1-amd64 #1 SMP Debian 5.8.10-2parrot1 (2020-10-05) x86_64 GNU/Linux

┌─[root@parrot]─[/home/an/Desktop/wfs]
└──╼ #ethtool -i wlan0

driver: rtl88XXau
version: 5.8.0-2parrot1-amd64
firmware-version: 
expansion-rom-version: 
bus-info: 2-1.1:1.0
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

┌─[root@parrot]─[/home/an/Desktop/wfs]
└──╼ #./Monitor.sh

PHY	Interface	Driver		Chipset

phy0	wlx00c0caac296e	88XXau		Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac

Killing these processes:

    PID Name
    710 wpa_supplicant

Interface wlan0
	ifindex 4
	wdev 0x1
	addr xx:xx:xx:xx:xx:xx
	type monitor
	wiphy 0
	txpower 30.00 dBm

┌─[root@parrot]─[/home/an/Desktop/wfs]
└──╼ #iwconfig

lo        no wireless extensions.

eth0      no wireless extensions.

eth1      no wireless extensions.

wlan0  unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

vmnet1    no wireless extensions.

vmnet8    no wireless extensions.

└──╼ #aireplay-ng --deauth 5 -a xx:xx:xx:xx:xx:xx -c xx:xx:xx:xx:xx:xx wlan0
10:18:20  Waiting for beacon frame (BSSID: xx:xx:xx:xx:xx:xx) on channel 6
10:18:20  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
10:18:21  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
10:18:21  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
10:18:22  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]
10:18:23  Sending 64 directed DeAuth (code 7). STMAC: [xx:xx:xx:xx:xx:xx] [ 0| 0 ACKs]

WPA handshake: xx:xx:xx:xx:xx:xx

harrysummer, 2021-01-19:
I am using Netgear A7000 on Raspberry Pi 4b with OpenWRT. And USB 3.0 works for me after setting `rtw_switch_usb_mode=1`. Not sure about injection. I am just using it as AP. With some code changes for backported mac80211, I was able to compile the code with make variables `CONFIG_PLATFORM_I386_PC=n CONFIG_PLATFORM_ARM64=y` and CFLAGS `-DRTW_P2P_GROUP_INTERFACE=1 -DRTW_DEDICATED_P2P_DEVICE=1`.

/etc/modules.d/rtl8814au:
```
8814au rtw_vht_enable=2 rtw_switch_usb_mode=1 rtw_power_mgnt=0 rtw_led_ctrl=0
```

Result of `lsusb -t`:
```
root@OpenWrt:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=, Driver=8814au, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=hub/4p, 480M
        |__ Port 4: Dev 4, If 0, Class=, Driver=usbhid, 12M
        |__ Port 4: Dev 4, If 1, Class=, Driver=usbhid, 12M
        |__ Port 4: Dev 4, If 2, Class=, Driver=usbhid, 12M
```

The only problem is that I can only use channel 36 in 802.11ac 5GHz range. And I need to reboot after setting to make it work.

ycycn, 2021-11-19:
at least one year passed
stupid

ipaqmaster, 2022-11-11:
Some highly unscientific research on my side but some experiments and thoughts:

This testing is on my `0846:9054 NetGear, Inc. Nighthawk A7000 802.11ac Wireless Adapter AC1900 [Realtek 8814AU]` USB WiFi card.

Works completely fine in USB2 mode, no complaints in any single way on this driver. None. (Yet)

USB3 mode however, **works entirely fine in 5GHz ranges** but "basically" not at all in the 2.4GHz space, with heavy exception. For example:

1. Setting the 88XXau driver's rtw_switch_usb_mode parameter to `1` and connecting the dongle works for USB3 mode, verifiable with `lsusb -t` showing it as a 5000M device (Keep an eye out that the parent device is also 5000M, 10000M or greater for the true USB 3.0 experience)

2. Setting the device to monitor mode as usual and using the classic `airodump-ng` with no extra args to scan the 2.4GHz range finds..... nothing. No APs.

3. If I turn on my iPhone's hotspot and check "Compatibility mode" (To force it to use 2.4G), the dongle can see this. Not too surprising as my phone is right here on the same desk. But the dongle does not see beacons from the other 20+ 2.4GHz APs nearby

This is using a very short 15cm braided Alogic USB-C>USB-A cable as my Dell XPS 13 here only takes USB-C.

-----------

Something that really started to capture my attention was [this paper from Intel regarding noise from the USB 3.0 standard and how it's (You guessed it) output in the 2.4-2.5GHz spectrum.
](https://www.usb.org/sites/default/files/327216.pdf)

So in a pathetic attempt and lack of better insulating material at hand... I tried grasping this tiny cable in my hands and the base of the USB adapter to (Very poorly) isolate it...and it worked. Kind of. The beacons for about 9 nearby BSSIDs started showing up which is not nearly as many as there are nearby, but a significant (infinite) improvement over the zero it can usually see (excluding my phone right here). Then I let go and the beacon counter stopped going up.

This leaves me thinking if NETGEAR really thought this thing deserved a USB3 while shipping it with a long magnetic USB3 extension cradle... they must have known 2.4G performance would've been horrible if it actually negotiates into USB3 mode as it seems USB3 and 2.4G on this unit don't seem to mix, not even a little.

I can only assume the `rtw_switch_usb_mode` is reserved explicitly for when dealing with 5GHz networks to take advantage of potential throughput if you somehow pull off a 40/80MHz wide channel without neighbor noise on your AP. (Only a best guess)

Unfortunately, [this paper from Intel](https://www.usb.org/sites/default/files/327216.pdf) seems to at least partially confirm why my poor-mans arm-cover test seemed to yield some results confirming that USB3 does interfere with the 2.4G range and in fact, very badly. At least with this dongle's design even when fully extended.

With that in mind, I can only assume USB3 mode is only intended for working with 5GHz activity and maybe the official driver switches on demand, or something.

---------------

TL;DR: https://www.usb.org/sites/default/files/327216.pdf the poor 2.4GHz reception on my USB Netgear A7000 could be expected in 88XXau's USB3 mode.

I'd love to try this again with a more expensive highly isolated USB3 cable or directly into a desktop motherboard to compare how it handles the noise for 2.4G reception

It's also likely that other USB3 wifi dongle manufactures know of this issue and may have isolated/designed their onboard components and antennas differently to help compensate. Not sure. Might be worth modding my A7000 here to help further isolate this later to see if I can avoid flipping between USB2/3 mode and just leave it in 3. Could be fun.

If this could still be a software level issue that doesn't involve just falling back to USB2 mode and "Dealing with it". I'd love to hear other experiences with other monitor-mode-capable USB3 WiFi dongles.