aircrack-ng/rtl8814au #3: DKMS build error on Debian 10

hardwareadictos, 2020-10-19:


kimocoder, 2020-10-19:
This is not a proper issue report. Need for info.. any info/output

hardwareadictos, 2020-10-19:
> This is not a proper issue report. Need for info.. any info/output

Sorry, just published it accidentally before making the report

kimocoder, 2020-10-19:
Your missing kmodsign.
More information here https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=939392

However, I'll add another DKMS install option without the kmodsign dependency shortly. I'll leave a note when done.

hardwareadictos, 2020-10-19:
> Your missing kmodsign.
> More information here https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=939392
> 
> However, I'll add another DKMS install option without the kmodsign dependency shortly. I'll leave a note when done.

Thanks a lot :)

Let me know if you want me to test it first

hardwareadictos, 2020-10-19:
Managed to delete kmodsign references on code, recompiled and driver just got installed:

root@atomikprometeus-dbn01:~# grep -ir kmodsign rtl8814au
rtl8814au/debian/changelog:  * add kmodsign
rtl8814au/debian/changelog.in:  * add kmodsign
rtl8814au/debian/rtl8814au-dkms.dkms.debhelper:MAKE="'make' clean; 'make' all;kmodsign sha512 MOK.priv MOK.der 8814au.ko"
rtl8814au/debian/rtl8814au-dkms.dkms:MAKE="'make' clean; 'make' all;kmodsign sha512 MOK.priv MOK.der 8814au.ko"
rtl8814au/debian/rtl8814au-dkms/usr/src/rtl8814au-5.8.5.1/dkms.conf:MAKE="'make' clean; 'make' all;kmodsign sha512 MOK.priv MOK.der 8814au.ko"

root@atomikprometeus-dbn01:~# dpkg -i rtl8814au-dkms_5.8.5.1_all.deb
(Reading database ... 102117 files and directories currently installed.)
Preparing to unpack rtl8814au-dkms_5.8.5.1_all.deb ...

------------------------------
Deleting module version: 5.8.5.1
completely from the DKMS tree.
------------------------------
Done.
Unpacking rtl8814au-dkms (5.8.5.1) over (5.8.5.1) ...
Setting up rtl8814au-dkms (5.8.5.1) ...
Loading new rtl8814au-5.8.5.1 DKMS files...
Building for 5.4.65-1-pve
Building initial module for 5.4.65-1-pve
Done.

8814au.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.65-1-pve/updates/dkms/

depmod...

DKMS: install completed.

Seems its on system:

root@atomikprometeus-dbn01:~# lsmod | grep 88
8814au               2166784  0
cfg80211              704512  1 8814au
btrfs                1241088  0
raid6_pq              114688  4 async_pq,btrfs,raid456,async_raid6_recov

But no leds powering up or interface detected on Debian. Maybe i'm missing something, not an expert on Wireless LAN drivers :P

kimocoder, 2020-10-19:
Ok. I also updated the DKMS install script,

$ make dkms_install
and
$ make dkms_remove

Is the commands.

closed 2020-10-19 by kimocoder