aircrack-ng/rtl8814au #4: Error while installing dkms deb package

AshwinRaam, 2020-10-19:
<pre><font color="#4E9A06"><b></b></font>:<font color="#3465A4"><b>~</b></font>$ sudo dpkg -i rtl8814au-dkms_5.8.5.1_all.deb 
Selecting previously unselected package rtl8814au-dkms.
(Reading database ... 187183 files and directories currently installed.)
Preparing to unpack rtl8814au-dkms_5.8.5.1_all.deb ...
Unpacking rtl8814au-dkms (5.8.5.1) ...
Setting up rtl8814au-dkms (5.8.5.1) ...
Loading new rtl8814au-5.8.5.1 DKMS files...
Building for 5.4.0-51-generic
Building initial module for 5.4.0-51-generic
ERROR (dkms apport): unable to determine source package for rtl8814au-dkms
Error! Bad return status for module build on kernel: 5.4.0-51-generic (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
<b>dpkg:</b> error processing package rtl8814au-dkms (--install):
 installed rtl8814au-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 rtl8814au-dkms
</pre>
[make.log](https://github.com/aircrack-ng/rtl8814au/files/5404377/make.log)
I have also attached the make log
Help me please.

kimocoder, 2020-10-19:
Do "git pull" in your 8814au folder.
Then do "make dkms_install"

Report back

AshwinRaam, 2020-10-19:
<pre>sudo make dkms_install
mkdir -p /usr/src/8814au-5.8.5.1
cp -r * /usr/src/8814au-5.8.5.1
dkms add -m 8814au -v 5.8.5.1

Creating symlink /var/lib/dkms/8814au/5.8.5.1/source -&gt;
                 /usr/src/8814au-5.8.5.1

DKMS: add completed.
dkms build -m 8814au -v 5.8.5.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
&apos;make&apos; -j4 KVER=5.4.0-51-generic KSRC=/lib/modules/5.4.0-51-generic/build...........
Signing module:
 - /var/lib/dkms/8814au/5.8.5.1/5.4.0-51-generic/x86_64/module/8814au.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.
dkms install -m 8814au -v 5.8.5.1

8814au.ko:
Running module version sanity check.

Good news! Module version 5.8.5.1 for 8814au.ko
exactly matches what is already found in kernel 5.4.0-51-generic.
DKMS will not replace this module.
You may override by specifying --force.

depmod....

DKMS: install completed.
dkms status
8814au, 5.8.5.1, 5.4.0-51-generic, x86_64: installed (WARNING! Diff between built and installed module!)
rtl8814au, 5.8.5.1: added</pre>
Yeah thank you so much i think it is installed now.
And do add this libelf-dev dependency in read-me because i faced an unknown error without it, whole day today, where, my make install is successful in ubuntu 20.04 but my new archer t9uh v2 didn't show up neither in ifconfig nor in iwconfig.

kimocoder, 2020-10-20:
Give me output from "lsusb" possible I need to add your VID/PID

AshwinRaam, 2020-10-20:
<pre><font color="#3465A4"><b>~</b></font>$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2357:0106 TP-Link Archer T9UH v1 [Realtek RTL8814AU]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
</pre>
This is my lsusb output. i don't know why it is coming v1 here but in the sticker on the adapter it is specified as v2 clearly.

D4rkiiee, 2020-12-18:
> sudo make dkms_install
> mkdir -p /usr/src/8814au-5.8.5.1
> cp -r * /usr/src/8814au-5.8.5.1
> dkms add -m 8814au -v 5.8.5.1
> 
> Creating symlink /var/lib/dkms/8814au/5.8.5.1/source ->
>                  /usr/src/8814au-5.8.5.1
> 
> DKMS: add completed.
> dkms build -m 8814au -v 5.8.5.1
> 
> Kernel preparation unnecessary for this kernel.  Skipping...
> 
> Building module:
> cleaning build area...
> 'make' -j4 KVER=5.4.0-51-generic KSRC=/lib/modules/5.4.0-51-generic/build...........
> Signing module:
>  - /var/lib/dkms/8814au/5.8.5.1/5.4.0-51-generic/x86_64/module/8814au.ko
> Secure Boot not enabled on this system.
> cleaning build area...
> 
> DKMS: build completed.
> dkms install -m 8814au -v 5.8.5.1
> 
> 8814au.ko:
> Running module version sanity check.
> 
> Good news! Module version 5.8.5.1 for 8814au.ko
> exactly matches what is already found in kernel 5.4.0-51-generic.
> DKMS will not replace this module.
> You may override by specifying --force.
> 
> depmod....
> 
> DKMS: install completed.
> dkms status
> 8814au, 5.8.5.1, 5.4.0-51-generic, x86_64: installed (WARNING! Diff between built and installed module!)
> 
> Yeah thank you so much i think it is installed now.
> And do add this libelf-dev dependency in read-me because i faced an unknown error without it, whole day today, where, my make install is successful in ubuntu 20.04 but my new archer t9uh v2 didn't show up neither in ifconfig nor in iwconfig.

Having the same issue:
 
`philip@minipc:~/rtl8814au/rtl8814au$ sudo make dkms_install 
mkdir -p /usr/src/8814au-5.8.5.1
cp -r * /usr/src/8814au-5.8.5.1
dkms add -m 8814au -v 5.8.5.1

Creating symlink /var/lib/dkms/8814au/5.8.5.1/source ->
                 /usr/src/8814au-5.8.5.1

DKMS: add completed.
dkms build -m 8814au -v 5.8.5.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j2 KVER=5.3.0-40-generic KSRC=/lib/modules/5.3.0-40-generic/build........................................................................
Signing module:
 - /var/lib/dkms/8814au/5.8.5.1/5.3.0-40-generic/x86_64/module/8814au.ko
Secure Boot not enabled on this system.
cleaning build area...

DKMS: build completed.
dkms install -m 8814au -v 5.8.5.1

8814au:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.3.0-40-generic/updates/dkms/

depmod....

DKMS: install completed.
dkms status
8814au, 5.8.5.1, 5.3.0-40-generic, x86_64: installed
rtl8814au, 5.8.5.1: added
philip@minipc:~/rtl8814au/rtl8814au$ iwconfig
enp1s0    no wireless extensions.

lo        no wireless extensions.

wlx00c0ca81a86f  IEEE 802.11  ESSID:"FRITZi"  
          Mode:Managed  Frequency:2.412 GHz  Access Point: C8:0E:14:45:F8:D2   
          Bit Rate=54 Mb/s   Tx-Power=20 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=70/70  Signal level=-23 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:371  Invalid misc:1564   Missed beacon:0

philip@minipc:~/rtl8814au/rtl8814au$ ifconfig
enp1s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 74:d4:35:8a:94:27  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 3932  bytes 422358 (422.3 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3932  bytes 422358 (422.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx00c0ca81a86f: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.178.72  netmask 255.255.255.0  broadcast 192.168.178.255
        inet6 fd00::6d5b:dab1:4671:73bc  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::5383:7749:7e61:ff91  prefixlen 64  scopeid 0x20<link>
        inet6 fd00::54c8:a127:73d2:3a4a  prefixlen 64  scopeid 0x0<global>
        ether 00:c0:ca:81:a8:6f  txqueuelen 1000  (Ethernet)
        RX packets 134398  bytes 168714057 (168.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 86659  bytes 9523661 (9.5 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

philip@minipc:~/rtl8814au/rtl8814au$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 004: ID 0bda:8187 Realtek Semiconductor Corp. RTL8187 Wireless Adapter
Bus 001 Device 003: ID 0bda:8813 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
philip@minipc:~/rtl8814au/rtl8814au$ 
`

I am just see my old Alfa card but not the new one. Can anybody help?