aircrack-ng/rtl8814au #5: Can't connect to 2.4 or 5G wireless networks. Wrong password issue...

nadyaeridhova, 2020-10-20:
Hi I have tried to connect to 4 different SSID's ( 2 different routers / 2 different SSIDs on both ) with no success. Network manager simply keeps asking for the password. I am running different linux distros on Virtualbox and attached ALFA Awus1900 device via USB 2.0 ( virtual interface ). Having the same issue On Linux Mint 19.3, Kali Linux 2020.3 and ParrotSec... Please find below the outputs from ParrotSec... BTW, there is no issue with the device itself as I have already successfully connected to all wireless SSID's from Windows 10 ( the host ).

#> uname -r
5.8.0-2parrot1-amd64

#> lsmod | grep 8814
8814au               2314240  0
cfg80211              892928  1 8814au
usbcore               323584  6 8814au,ohci_hcd,ehci_pci,usbhid,ehci_hcd,ohci_pci

#> lsusb
Bus 001 Device 002: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

# --- /var/log/syslog --- #

Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.0988] Config: added 'ssid' value '[MYSSID]'
Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.0988] Config: added 'scan_ssid' value '1'
Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.0988] Config: added 'bgscan' value 'simple:30:-70:86400'
Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.0988] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.0988] Config: added 'auth_alg' value 'OPEN'
Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.0988] Config: added 'psk' value '<hidden>'
Oct 20 17:59:24 parrot NetworkManager[640]: <info>  [1603205964.1081] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: inactive -> scanning
Oct 20 17:59:30 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: Trying to associate with [MACOFMYAP] (SSID='[MYSSID]' freq=2437 MHz)
Oct 20 17:59:30 parrot wpa_supplicant[652]: nl80211: kernel reports: Authentication algorithm number required
Oct 20 17:59:30 parrot NetworkManager[640]: <info>  [1603205970.3980] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: scanning -> associating
Oct 20 17:59:30 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2437 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2437 cf2=0
Oct 20 17:59:37 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-ASSOC-REJECT status_code=1
Oct 20 17:59:37 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Oct 20 17:59:37 parrot NetworkManager[640]: <info>  [1603205977.3005] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: associating -> disconnected
Oct 20 17:59:37 parrot NetworkManager[640]: <info>  [1603205977.4079] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: disconnected -> scanning
Oct 20 17:59:41 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: Trying to associate with [MACOFMYAP] (SSID='[MYSSID]' freq=2437 MHz)
Oct 20 17:59:41 parrot wpa_supplicant[652]: nl80211: kernel reports: Authentication algorithm number required
Oct 20 17:59:41 parrot NetworkManager[640]: <info>  [1603205981.0418] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: scanning -> associating
Oct 20 17:59:42 parrot systemd[1]: systemd-hostnamed.service: Succeeded.
Oct 20 17:59:49 parrot NetworkManager[640]: <warn>  [1603205989.2069] device (wlx[PERMADDROFMYDEVICE]): Activation: (wifi) association took too long
Oct 20 17:59:49 parrot NetworkManager[640]: <info>  [1603205989.2069] device (wlx[PERMADDROFMYDEVICE]): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Oct 20 17:59:49 parrot NetworkManager[640]: <warn>  [1603205989.2079] device (wlx[PERMADDROFMYDEVICE]): Activation: (wifi) asking for new secrets
Oct 20 17:59:49 parrot dbus-daemon[639]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.42' (uid=1000 pid=2784 comm="nm-applet ")
Oct 20 17:59:49 parrot systemd[1]: Starting Hostname Service...
Oct 20 17:59:50 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-DISCONNECTED bssid=[MACOFMYAP] reason=3 locally_generated=1
Oct 20 17:59:50 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-ASSOC-REJECT status_code=1
Oct 20 17:59:50 parrot NetworkManager[640]: <info>  [1603205990.2328] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: associating -> disconnected
Oct 20 17:59:50 parrot dbus-daemon[639]: [system] Successfully activated service 'org.freedesktop.hostname1'
Oct 20 17:59:50 parrot systemd[1]: Started Hostname Service.
Oct 20 18:00:20 parrot systemd[1]: systemd-hostnamed.service: Succeeded.
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3158] device (wlx[PERMADDROFMYDEVICE]): state change: need-auth -> prepare (reason 'none', sys-iface-state: 'managed')
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3160] device (wlx[PERMADDROFMYDEVICE]): state change: prepare -> config (reason 'none', sys-iface-state: 'managed')
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] device (wlx[PERMADDROFMYDEVICE]): Activation: (wifi) connection '[MYSSID]' has security, and secrets exist.  No new secrets needed.
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] Config: added 'ssid' value '[MYSSID]'
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] Config: added 'scan_ssid' value '1'
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] Config: added 'bgscan' value 'simple:30:-70:86400'
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256 FT-PSK'
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] Config: added 'auth_alg' value 'OPEN'
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3162] Config: added 'psk' value '<hidden>'
Oct 20 18:00:48 parrot NetworkManager[640]: <info>  [1603206048.3253] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: disconnected -> scanning
Oct 20 18:00:50 parrot systemd[1]: pcscd.service: Succeeded.
Oct 20 18:00:54 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: Trying to associate with [MACOFMYAP] (SSID='[MYSSID]' freq=2437 MHz)
Oct 20 18:00:54 parrot wpa_supplicant[652]: nl80211: kernel reports: Authentication algorithm number required
Oct 20 18:00:54 parrot NetworkManager[640]: <info>  [1603206054.6172] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: scanning -> associating
Oct 20 18:00:54 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=2437 ht_enabled=1 ch_offset=0 ch_width=20 MHz cf1=2437 cf2=0
Oct 20 18:01:01 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-ASSOC-REJECT status_code=1
Oct 20 18:01:01 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Oct 20 18:01:01 parrot NetworkManager[640]: <info>  [1603206061.5235] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: associating -> disconnected
Oct 20 18:01:02 parrot NetworkManager[640]: <info>  [1603206062.0297] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: disconnected -> scanning
Oct 20 18:01:05 parrot wpa_supplicant[652]: wlx[PERMADDROFMYDEVICE]: Trying to associate with [MACOFMYAP] (SSID='[MYSSID]' freq=2437 MHz)
Oct 20 18:01:05 parrot wpa_supplicant[652]: nl80211: kernel reports: Authentication algorithm number required
Oct 20 18:01:05 parrot NetworkManager[640]: <info>  [1603206065.5834] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: scanning -> associating
Oct 20 18:01:13 parrot NetworkManager[640]: <warn>  [1603206073.2052] device (wlx[PERMADDROFMYDEVICE]): Activation: (wifi) association took too long
Oct 20 18:01:13 parrot NetworkManager[640]: <info>  [1603206073.2052] device (wlx[PERMADDROFMYDEVICE]): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Oct 20 18:01:13 parrot NetworkManager[640]: <warn>  [1603206073.2059] device (wlx[PERMADDROFMYDEVICE]): Activation: (wifi) asking for new secrets
Oct 20 18:01:13 parrot systemd[1]: Started PC/SC Smart Card Daemon.
[... cut ...]
Oct 20 18:01:53 parrot NetworkManager[640]: <info>  [1603206113.6549] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: disconnected -> interface_disabled
Oct 20 18:01:53 parrot NetworkManager[640]: <info>  [1603206113.6564] device (wlx[PERMADDROFMYDEVICE]): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Oct 20 18:01:53 parrot NetworkManager[640]: <info>  [1603206113.6965] device (wlx[PERMADDROFMYDEVICE]): supplicant interface state: interface_disabled -> disconnected

If you need any other logs or information, I would like to share them also...

nadyaeridhova, 2020-10-24:
Well, I have just figured out that when I attached the ALFA Awus1900 via "USB 3.0 Controller" to VirtualBox. The password loop issue has gone away. I don't know if this issue has anything to do with the driver itself. So I am closing it...

closed 2020-10-24 by nadyaeridhova

tmccartney, 2020-10-26:
I seem to be having the same/similar issue (password loop). 
I followed the modprobe method to change from USB 2.0 (Default?) to USB 3.0 & it resolved the problem & connected successfully. 

Running Manjaro 5.8 kernel, installed with dkms method.