aircrack-ng/rtl8814au #6: Channel -1 and injection didn't work

AbdelrhmanKhater, 2020-10-21:
> With airodump-ng, I got fixed channel wlan0: -1  
airodump-ng --bssid xx:xx:xx:xx:xx:xx-c x wlan0
[image]
With aireplay-ng, I can't use it without switch --ignore-negative-one 
aireplay-ng -0 0 -a xx:xx:xx:xx:xx:xx -c xx:xx:xx:xx:xx:xx --ignore-negative-one wlan0
[image]
Injection failed

AshwinRaam, 2020-10-21:
Yeah even i am facing the exact same problem.

kimocoder, 2020-10-21:
Injection is down atm, needed to fix some kernel taint and other issues. Will fix it when I get back from work.

AshwinRaam, 2020-10-21:
yeah waiting for it kim.

AbdelrhmanKhater, 2020-10-21:
> Injection is down atm, needed to fix some kernel taint and other issues. Will fix it when I get back from work.

Nice, I'm waiting for your update.

Dukezo, 2020-10-24:
I'm having the same issue. Any news regarding the fix?

AbdelrhmanKhater, 2020-10-26:
> I'm having the same issue. Any news regarding the fix?

Not yet

AshwinRaam, 2020-10-26:
Damn i wish i could find some good resource to create wifi driver, because i can't wait to test my new adapter's capability. Can someone suggest me if any is available. I would like to contribute to this code. @kimocoder

desegura, 2020-11-06:
+1 can we help you @kimocoder ? We are excited to test the power of AWUS1900 with Aircrack-ng ☺

apteryxx, 2020-11-13:
+1. I've install drivers from this repo, and injection doesn't work anyway with the same error message. With older drivers, Alpha AC1900 working not good, but injection is worked.

skiocink, 2020-11-14:
I'm still on 5.6.4.2 but waiting with so much interest to next driver version for 8814

johnconway11, 2020-11-22:
+1 for me too, hopefully won't be too long before the issue is fixed

paleknight894, 2020-12-01:
> Injection is down atm, needed to fix some kernel taint and other issues. Will fix it when I get back from work.

do you mean Asynchronous Transfer Mode in kernel configuration?

tr4g, 2021-01-13:
3 months later, no news? i thought we was gonna do it when he came back from work lol

Nj0n, 2021-01-27:
yea same problem here too. is there any news?