aircrack-ng/rtl8814au #7: Raspberry Pi support

leow149, 2020-10-21:
What about the good old Raspi support? Will it get added?

raplin, 2020-10-23:
fixed in a PR I just submitted (and TX power control) - works on a Pi4 again now

Excelsior-Ink, 2020-10-27:
I tried the fork on my RPi 4 on the latest RPi OS 64-bit. I did:

sudo apt install --reinstall raspberrypi-kernel-headers raspberrypi-kernel
reboot
sudo apt install git
git clone https://github.com/raplin/rtl8814au.git
make dkms_install ARCH=arm64

and I get:

Building module:
cleaning build area...
'make' -j4 KVER=5.4.72-v8+ KSRC=/lib/modules/5.4.72-v8+/build...........................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.4.72-v8+ (aarch64)
Consult /var/lib/dkms/8814au/5.8.5.1/build/make.log for more information.
make: *** [Makefile:2410: dkms_install] Error 10

cat /var/lib/dkms/8814au/5.8.5.1/build/make.log

  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/8814au/5.8.5.1/build/8814au.o
  Building modules, stage 2.
  MODPOST 1 modules
ERROR: "__stack_chk_guard" [/var/lib/dkms/8814au/5.8.5.1/build/8814au.ko] undefined!
make[3]: *** [scripts/Makefile.modpost:94: __modpost] Error 1
make[2]: *** [Makefile:1645: modules] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.4.72-v8+'
make[1]: *** [Makefile:2353: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8814au/5.8.5.1/build'

Any idea?

raplin, 2020-10-28:
Can't help you on 64-bit, I am running 32-bit kernel; 
compare the Makefile and build steps to the one in https://github.com/aircrack-ng/rtl8812au - my guess there is probably some minor 64-bit issue in the Makefile in this fork

Excelsior-Ink, 2020-10-28:
Thank you! Your update was helpful. I have noticed that the key error is

ERROR: "__stack_chk_guard" [/var/lib/dkms/8814au/5.8.5.1/build/8814au.ko] undefined!

So I have created a new section for Raspberry Pi 64-bit as follows:

ifeq ($(CONFIG_PLATFORM_ARM64_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
EXTRA_CFLAGS += -fno-stack-protector
ARCH := arm64
CROSS_COMPILE ?=
KVER ?= $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
INSTALL_PREFIX :=
endif

They key was adding EXTRA_CFLAGS += -fno-stack-protector and it compiled.
I have no idea what this does, so it might be a temporary fix.

raplin, 2020-10-28:
thanks, will fix at some point

Excelsior-Ink, 2020-10-30:
One extra line is needed:

EXTRA_CFLAGS += -fno-stack-protector

TBH I don't know what it does, I am no C programmer. But without this line compiling fails

ndrwdslv, 2020-10-30:
> One extra line is needed:
> 
> EXTRA_CFLAGS += -fno-stack-protector
> 
> TBH I don't know what it does, I am no C programmer. But without this line compiling fails

Weird, I didn't need it for Raspberry Pi OS Buster on Rasp Pi 4. 

Pretty sure yours failed because you didn't do:
sed -i 's/^\tdkms build/\tARCH=arm dkms build/' Makefile
sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf

Because your traceback/error doesn't mention it before the 'make':
'make' -j4 KVER=5.4.72-v8+ KSRC=/lib/modules/5.4.72-v8+/build...........................................(bad exit status: 2)

Mind doing a dkms_remove and try installing again, but with the ARCH=arm in your Makefile/dkms.conf?

Excelsior-Ink, 2020-10-30:
I did it, it fails. The difference is I am using 64-bit https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip

ndrwdslv, 2020-10-30:
> I did it, it fails. The difference is I am using 64-bit https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2020-08-24/2020-08-20-raspios-buster-arm64-lite.zip

Weird.. I am too.. damn, I use a clean install though, oh well.

rjbrown99, 2020-12-07:
> Looks like you forgot to run the following command just like on rtl8812au:
> `sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf`
> 
> Not sure if that could also cause that crash.
> 
> Should be noted that the same steps for Rasp Pi should be done as rtl8812au: (arm64 isn't an option, stick to arm)
> `sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile`
> `sed -i 's/^\tdkms build/\tARCH=arm dkms build/' Makefile`
> `sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf`
> 
> Ctrl-F "CONFIG_PLATFORM_I386_PC" and add the following under it:
> `CONFIG_PLATFORM_ARM64_RPI = y`
> 
> Go to line 1290 and add this block:
> `ifeq ($(CONFIG_PLATFORM_ARM64_RPI), y)`
> `EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN`
> `EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT`
> `ARCH ?= arm64`
> `CROSS_COMPILE ?=`
> `KVER ?= $(shell uname -r)`
> `KSRC := /lib/modules/$(KVER)/build`
> `MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/`
> `INSTALL_PREFIX :=`
> `endif`

FWIW, I'm now using this PR for the 64 bit version of Raspberry Pi OS. The original PR plus the ideas above did the trick.

Think3wang, 2020-12-23:
Does this driver works on Pi 3 B+ ? Thanks.

withernet, 2021-01-11:
I believe you can close this; I was able to get the module to build by setting installing headers on the pi 4, then `export ARCH=arm`, and following the rest of the build process.

However, with that being said, injection is still broken, see: https://github.com/aircrack-ng/rtl8814au/issues/6

For reference:

```
$ uname -r
5.4.83-v7l+

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

$ lsmod | grep 8814
8814au               1978368  0
cfg80211              675840  2 8814au,brcmfmac
```

joaquinabian, 2021-02-05:
> I believe you can close this

Please, add the working instructions for installing the driver on a Raspi4 on the README.md before closing this thread.

funjspi, 2021-07-26:
I am trying to get a tp-link AC600 (archer T2U Plus) wifi dongle working on my PI 3 B to no avail. I have tried everything here. is this broken?

closed 2025-03-02 by leow149