aircrack-ng/rtl8814au #9: Error while setting up rtl88184au-dkms

tomirio619, 2020-10-27:
I experience the same problem as described in #4.
After a successful `make install`, I run `sudo apt-get upgrade`, which gives the following output:

```
Setting up rtl8814au-dkms (5.8.5.1) ...
Loading new rtl8814au-5.8.5.1 DKMS files...
Building for 5.4.0-52-generic
Building initial module for 5.4.0-52-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/rtl8814au-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.4.0-52-generic (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
dpkg: error processing package rtl8814au-dkms (--configure):
 installed rtl8814au-dkms package post-installation script subprocess returned error exit status 10
```

The [make.log](https://github.com/aircrack-ng/rtl8814au/files/5443964/make.log)

Relevant `lsusb` output:

```
Bus 003 Device 002: ID 0b05:1853 ASUSTek Computer, Inc. 802.11ac NIC
```

I'm not using secure boot:
```
# mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode
```

Do you have any ideas?

AshwinRaam, 2020-10-30:
Yeah use sudo make dkms_install to install the drivers it will work @tomirio619.

tomirio619, 2020-12-10:
I think I was mixing the installations, It is working now

closed 2020-12-10 by tomirio619

Chal13W1zz, 2021-12-05:
> I think I was mixing the installations, It is working now

 Hello, @tomirio619 . How did you solve this?