aircrack-ng/rtl8814au #10: Cannot turn off the LED

MR-Diamond, 2020-11-10:
Hi, I switched from 88XXau to 8814au, for TP-link Archer T9UH;
previously I was able to turn off the led by using `rtw_led_ctrl=0` now instead this option is no longer available, or at least I don't see any related.
Is to be implemented?

gebau00a, 2021-02-08:
I identified there's a file in /proc/net/8814au/[adaptername]/led_config
Writing into this file changes the led behavior, but i could not generate a standard output.
Device is D-Link DWA-192 

`sudo echo 0 > led_config`
turned the LED constantly on, no blinking anymore.

Writing whatever value to the file does not change the value of bRegUseLed which I assume the trigger the LED.

Before the drivers for rtl8814 and rtl8812 where separated, the LED_ctrl worked on this device, so it's solely a driver issue

gebau00a, 2021-02-10:
I created a fork and asked for a pull request to implement the feature [https://github.com/aircrack-ng/rtl8814au/pull/36](https://github.com/aircrack-ng/rtl8814au/pull/36)