aircrack-ng/rtl8814au #14: Can not use AP and 11ac

pdionisis, 2020-12-07:
I can not use AP mode with hostap at an ubuntu 20.04 with 11ac mode.
11n mode seems to work.

I use the create_ap
If i try to force 11ac it uses plain 11a mode(54mbps)

Card edup-1612 with 4 antennas , 3 streams

pdionisis, 2020-12-09:
I managed to enable 802.11ac at AP mode using the suggestion of :
https://github.com/aircrack-ng/rtl8812au/issues/272

sudo modprobe -r 8814au 
sudo modprobe 8814au rtw_vht_enable=2

I even managed to have a 500mbit transfer with edup 1612 (8814au) as server and an intel 8260 (2 streams) as client
very close to each other.

Now my problem is that 
1) the iw list returns channels that are not accepted (hostapd says that it can not use them and stops)
  Is it possible that nl80211 or hostapd have different embeded channels than the card ? (I tried setting to country to hostapd and also setting GR or US. 

2) When usinf channel 36 the speed is top 300mbit , when using 5740 (or something) it goes up to 500mbit (real tcp iperf3 speed).
At both times the connection speed is (and maintained at 866mbit).
Maybe it has something to do with neighbour bss scanning? 
At chan 36 there is another 5Ghz AP (with very low signal). At 5740 there is nothing
Is there a possibility that something is (silently) disabled because of the neighbouring AP?
Shouldn't this disablement be reported? (both times the connection speed is 866)

Any way to disable neighbouring bss scanning?

Any other idea for the low speed at 5180 (chan 36) versus 5740?

logan001, 2020-12-12:
thank u for the info about that parameter. now i can get 220 mbps (connection report 433mbps) with my phone instead of 110-115mbps without it.

now for the adapter i use is a asus USB-AC68 that has this same chip.
so for me this driver works great on 36-44 channels.  i don't have any laptop to test it better.