aircrack-ng/rtl8814au #16: System freeze after running AP

Ludo444, 2020-12-11:
Compilation and running of driver seems fine initially. However, after running `hostapd` for a few hours (it varies greatly), system freezes and is completely unresponsive. There's nothing in logs that could indicate what's going sideways. I had same issue with other published drivers based on version `5.8.5.1`.

The ones based on older version (`5.6.4.2`) like https://github.com/n0ss/realtek-rtl88xxau-dkms  are working fine.

Distribution: Debian 11
Tested kernels: 5.2, 5.4, 5.9

logan001, 2020-12-12:
so this is the culprit of my arch freezing.

andrejpodzimek, 2020-12-21:
I have exactly the same problem.

Hardware: Alfa `AWUS1900`, ASRock Creator x570 with Ryzen 3950X with [this ArchLinux package](https://aur.archlinux.org/packages/rtl8814au-aircrack-dkms-git/) and kernel `5.9.14`.

The freezes are reproducible, more or less: Flip your cell phone's WiFi on and off a couple of times (at most 10, usually) to make it (de)authenticate repeatedly and the whole system running `hostapd` will be dead frozen. This is _bad_ for a non-redundant home server, because there is:
* no kernel panic and therefore no automatic reboot,
* no reponse to any inputs, network or HID or magic SysRq,
* nothing logged via `dmesg`&rightarrow;`journalctl`; the world just stops for the entire machine
* (yet the SSID of the WiFi network is still visible for an extra while, a few minutes).

(Side note: There is also [another version of the driver](https://aur.archlinux.org/packages/rtl8814au-dkms-git/), but that's a no-go as well, because with that one your WiFi AP just stops working a couple of times a day and you have to unload and reload `8814au` to get it running again. (On the other hand, it won't freeze the whole machine.))

I don't set any wild module options, just this: `options 8814au rtw_switch_usb_mode=1`

For the record, here's the `dmesg` with device details. It's an Alfa USB adapter, type `AWUS1900` (and it does have a real S/N printed on it (`17BK1900U0028`), but the driver always says `123456`):

```
Dec 21 06:46:20 charon kernel: usb 9-1: new high-speed USB device number 9 using xhci_hcd
Dec 21 06:46:20 charon kernel: usb 9-1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
Dec 21 06:46:20 charon kernel: usb 9-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 21 06:46:20 charon kernel: usb 9-1: Product: 802.11ac NIC
Dec 21 06:46:20 charon kernel: usb 9-1: Manufacturer: Realtek
Dec 21 06:46:20 charon kernel: usb 9-1: SerialNumber: 123456
Dec 21 06:46:21 charon kernel: usb 9-1: USB disconnect, device number 9
Dec 21 06:46:21 charon kernel: usbcore: registered new interface driver 8814au
Dec 21 06:46:22 charon kernel: usb 10-1: new SuperSpeed Gen 1 USB device number 5 using xhci_hcd
Dec 21 06:46:22 charon kernel: usb 10-1: Int endpoint with wBytesPerInterval of 512 in config 1 interface 0 altsetting 0 ep 133: setting to 64
Dec 21 06:46:22 charon kernel: usb 10-1: New USB device found, idVendor=0bda, idProduct=8813, bcdDevice= 0.00
Dec 21 06:46:22 charon kernel: usb 10-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Dec 21 06:46:22 charon kernel: usb 10-1: Product: 802.11ac NIC
Dec 21 06:46:22 charon kernel: usb 10-1: Manufacturer: Realtek
Dec 21 06:46:22 charon kernel: usb 10-1: SerialNumber: 123456
```

Here's what's _not_ commented-out in my `hostapd.conf` (but many of those options are there by default, uncommented, so this dump is only meaningful if there's a particular culprit option that someone can spot):

```
interface=charonwifi1
driver=nl80211
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
ctrl_interface=/run/hostapd
ctrl_interface_group=0
ssid=Net4
country_code=CH
country3=0x49
ieee80211d=1
ieee80211h=1
hw_mode=a
channel=169
beacon_int=100
dtim_period=2
rts_threshold=-1
fragm_threshold=-1
macaddr_acl=1
accept_mac_file=/etc/hostapd/hostapd.accept
auth_algs=1
ignore_broadcast_ssid=0
wmm_enabled=1
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0
ieee80211n=1
ht_capab=[GF][HT40-][LDPC][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]
require_ht=1
obss_interval=300
ieee80211ac=1
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][BF-ANTENNA-3][SOUNDING-DIMENSION-4][HTC-VHT][MAX-A-MPDU-LEN-EXP7][MAX-MPDU-11454]
use_sta_nsts=1
ieee8021x=1
eapol_version=2
eapol_key_index_workaround=0
eap_server=1
eap_user_file=/etc/hostapd/hostapd.eap_user
ca_cert=/var/podzimek-ca/podzimek-ca-bundle.pem
server_cert=/etc/hostapd/hostapd.crt
private_key=/etc/hostapd/hostapd.key
check_crl=0
wpa=2
wpa_key_mgmt=WPA-EAP-SHA256
wpa_pairwise=CCMP
rsn_pairwise=CCMP
group_cipher=CCMP
wpa_group_rekey=3600
wpa_strict_rekey=1
wpa_gmk_rekey=86400
ieee80211w=2
ocv=1
ap_table_max_size=0
ap_table_expiration_time=0
tdls_prohibit=0
tdls_prohibit_chan_switch=0
time_advertisement=2
time_zone=CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
```

As already mentioned, there is _nothing_ WiFi-related in `dmesg` before the freeze. Sadly this machine doesn't have a serial console that could possibly capture stuff that wasn't logged due to the freeze and reproduction attempts under `netconsole` would require a separate setup on another machine, because once the server is down, so is my entire network and Internet presence and storage.

On the same machine, an Intel AX200-based WiFi AP with `hostapd` is rock solid stable (albeit with a catastrophic performance and 2.4 GHz-only operation; this is a [known issue](https://community.intel.com/t5/Wireless/Wi-Fi-6-AX200-cannot-open-the-5G-frequency-band-for-hotspot-use/td-p/1218738)).

jmstark, 2021-01-21:
Same happens on my machine. I used another driver that worked well, but since I upgraded my Ubuntu install, I had to switch the driver since the old one doesn't compile on newer kernels. When I run a hotspot, the machine freezes after a while.

logan001, 2021-02-14:
what i've found out is that my computer freeze whene one of my devices stop using the AP. i can have multiple devices connected to my AP but if one for whatever reason goes offline then there is 80% my computer will freeze. its not an instant freeze. it takes a few moments (probably up to 2-3 minutes) for computer to freeze.

vorticle, 2021-02-14:
I believe that this might be the same is thus bug here: [https://github.com/aircrack-ng/rtl8812au/issues/625](url)

I tried patching the source, we will see.

gusarg81, 2021-04-25:
Hi,

Does this got fixed? Same is happening to me with the driver from https://github.com/morrownr/8814au

jmstark, 2021-12-07:
Here's a possible fix: https://github.com/morrownr/8814au/pull/19
(thank you @vorticle, @gusarg81 and all Devs).
I didn't try it yet.

Btw the link in @vorticle's comment leads to a 404, while copy+pasting the text works. This is the correct link:
> I believe that this might be the same is this bug here: https://github.com/aircrack-ng/rtl8812au/issues/625

gusarg81, 2021-12-07:
Yeah, seems that fixed because I don't have that problem anymore and I use two devices at same time (one per band).

vorticle, 2022-01-10:
Oops, yes, you are right, I made a mistake with the link there. Thank you.

I can also confirm that this fixed the issue: my machine is running stable for months now.