aircrack-ng/rtl8814au #28: Not working on Ubuntu 20.04

MorganLee909, 2021-01-13:
I tried installing this on my machine and it wasn't working.  However, the older(?) version worked perfectly for me: https://github.com/aircrack-ng/rtl8812au.  I have no idea what the problem is, but I just wanted to bring it up in case it helps.  Here is some of my machine info:

OS: Ubuntu 20.04 (with KDE Plasma)
Kernel: 5.4.0
Adapter: TP-Link Archer T2U Plus

Let me know if you want other info from me.

DPlachkov, 2021-01-21:
@MorganLee909  I did manage to get it to work after installing kali vm -> update upgrade dist-upgrade, install linux headers, then make + make install and rebooting a few times, but the signal in kali seems weak and packet injection seems to not work as well. How does it work with the older rtl8812au driver?

kcdtv, 2021-01-22:
Hi there! I've just tried in Ubuntu 20.04:
```
Linux 5.4.0-64-generic #72-Ubuntu SMP Fri Jan 15 10:27:54 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```
I got a minor warning while i compiled: 
```
CC [M]  /home/kcdtv/WiFi/rtl8814au/os_dep/linux/recv_linux.o
  CC [M]  /home/kcdtv/WiFi/rtl8814au/os_dep/linux/ioctl_cfg80211.o
/home/kcdtv/WiFi/rtl8814au/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/home/kcdtv/WiFi/rtl8814au/os_dep/linux/ioctl_cfg80211.c:468:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  468 |     if (!rtw_cfg80211_allow_ch_switch_notify(adapter))
      |     ^~
/home/kcdtv/WiFi/rtl8814au/os_dep/linux/ioctl_cfg80211.c:470:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  470 |   cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/kcdtv/WiFi/rtl8814au/os_dep/linux/rtw_cfgvendor.o 
```
But nothing that prevented me to install the drivers.
when i plugged the awus1900 my desktop froze, i could move the mouse but that's it.... 
I had to reboot and it was impossible to reboot with the wifi device plugged in.
I got this error message: 
[image]
It cannot be an EFI issue as my old desktop does not have EFI but old school BIOS
So i tried to reboot without the WiFi device plunged-in, stopped network manager, plugged the device and then the desktop did not froze. 
`lsusb` recongnize the device but when i try to detect it with `ìw dev` my shell stay in the limbo... 
error message with dmesg:
```
[ 2538.724030] INFO: task iwd:767 blocked for more than 120 seconds.
[ 2538.724036]       Tainted: P           OE     5.4.0-64-generic #72-Ubuntu
[ 2538.724038] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2538.724040] iwd             D    0   767      1 0x00004100
[ 2538.724043] Call Trace:
[ 2538.724052]  __schedule+0x2e3/0x740
[ 2538.724055]  schedule+0x42/0xb0
[ 2538.724057]  schedule_preempt_disabled+0xe/0x10
[ 2538.724059]  __mutex_lock.isra.0+0x182/0x4f0
[ 2538.724062]  ? rcu_exp_wait_wake+0x120/0x120
[ 2538.724066]  ? device_del+0x2a1/0x370
[ 2538.724068]  __mutex_lock_slowpath+0x13/0x20
[ 2538.724070]  mutex_lock+0x2e/0x40
[ 2538.724073]  rtnl_lock+0x15/0x20
[ 2538.724107]  wiphy_unregister+0x40/0x340 [cfg80211]
[ 2538.724162]  rtw_wiphy_unregister+0x1a/0x1d [8814au]
[ 2538.724208]  rtw_os_ndev_unregister+0x75/0xbf [8814au]
[ 2538.724254]  cfg80211_rtw_del_virtual_intf+0x4d/0x9a [8814au]
[ 2538.724274]  nl80211_del_interface+0x4f/0x110 [cfg80211]
[ 2538.724277]  genl_family_rcv_msg+0x1b9/0x470
[ 2538.724280]  genl_rcv_msg+0x4c/0xa0
[ 2538.724282]  ? _cond_resched+0x19/0x30
[ 2538.724284]  ? genl_family_rcv_msg+0x470/0x470
[ 2538.724286]  netlink_rcv_skb+0x50/0x120
[ 2538.724288]  genl_rcv+0x29/0x40
[ 2538.724290]  netlink_unicast+0x187/0x220
[ 2538.724292]  netlink_sendmsg+0x222/0x3e0
[ 2538.724295]  sock_sendmsg+0x65/0x70
[ 2538.724297]  __sys_sendto+0x113/0x190
[ 2538.724301]  ? __secure_computing+0x42/0xe0
[ 2538.724304]  ? syscall_trace_enter+0x134/0x2b0
[ 2538.724307]  __x64_sys_sendto+0x29/0x30
[ 2538.724309]  do_syscall_64+0x57/0x190
[ 2538.724311]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 2538.724313] RIP: 0033:0x7f4a5aefd690
[ 2538.724319] Code: Bad RIP value.
[ 2538.724321] RSP: 002b:00007ffd0f08adf8 EFLAGS: 00000246 ORIG_RAX: 000000000000002c
[ 2538.724323] RAX: ffffffffffffffda RBX: 00005639582cf9a0 RCX: 00007f4a5aefd690
[ 2538.724324] RDX: 0000000000000028 RSI: 00005639582e9740 RDI: 0000000000000004
[... 294 lines cut ...]
[ 2780.385317]  netlink_rcv_skb+0x50/0x120
[ 2780.385318]  rtnetlink_rcv+0x15/0x20
[ 2780.385320]  netlink_unicast+0x187/0x220
[ 2780.385322]  netlink_sendmsg+0x222/0x3e0
[ 2780.385324]  sock_sendmsg+0x65/0x70
[ 2780.385325]  __sys_sendto+0x113/0x190
[ 2780.385330]  ? fd_install+0x27/0x30
[ 2780.385332]  ? __sys_socket+0x9e/0xf0
[ 2780.385334]  __x64_sys_sendto+0x29/0x30
[ 2780.385335]  do_syscall_64+0x57/0x190
[ 2780.385337]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 2780.385338] RIP: 0033:0x7f18b02bc844
[ 2780.385340] Code: Bad RIP value.
[ 2780.385341] RSP: 002b:00007f188cbbc130 EFLAGS: 00000293 ORIG_RAX: 000000000000002c
[ 2780.385343] RAX: ffffffffffffffda RBX: 00007f188cbbd250 RCX: 00007f18b02bc844
[ 2780.385344] RDX: 0000000000000014 RSI: 00007f188cbbd250 RDI: 000000000000003f
[ 2780.385345] RBP: 0000000000000000 R08: 00007f188cbbd1f4 R09: 000000000000000c
[ 2780.385346] R10: 0000000000000000 R11: 0000000000000293 R12: 00007f188cbbd1f4
[ 2780.385347] R13: 00007f188cbbd880 R14: 00007f188cbbd210 R15: 000000000000003f
[ 3275.272226] usb 1-5: reset high-speed USB device number 5 using ehci-pci
[ 3279.340595] usb 1-5: USB disconnect, device number 5
```
And i cannot execute any order as  sudo, it freeze the superuser. 
Any suggestion? 
From what i understood this `Code: Bad RIP value.` suggests that mistake in compilation could be the cause of this issue... Here i am for any testing, don't hesitate to ask.

MorganLee909, 2021-01-23:
@DPlachkov @kcdtv I can't speak for any other computers, but the older driver worked perfectly.  I downloaded it, followed the instructions in the README, restarted the computer and then it just worked.  I did have a problem that it stopped working once, but I ran the install again, restarted again, and then it worked.  Link is in my first comment.

DPlachkov, 2021-01-24:
> @DPlachkov @kcdtv I can't speak for any other computers, but the older driver worked perfectly. I downloaded it, followed the instructions in the README, restarted the computer and then it just [...]

How is the AWUS1900 signal for you? It should be way better but I get the same signal strength as the TP link TL-WN722N, not only that but the 2nd one is Atheros and seems to inject way better

kcdtv, 2021-01-25:
> @DPlachkov @kcdtv I can't speak for any other computers, but the older driver worked perfectly. I downloaded it, followed the instructions in the README, restarted the computer and then it just [...]

Yes, the "older driver" worked for me, but the goal is to have the "new one" working too in most systems possible. if it can be.  

> How is the AWUS1900 signal for you? It should be way better but I get the same signal strength as the TP link TL-WN722N, not only that but the 2nd one is Atheros and seems to inject way better

The 2 chipsets have more or less the same sensibility so it is normal that you don't see any drastic change. 
We got around -95dBm in 2.4Ghz 10 years ago and that won't improve. If you are searching for a USB stick to audit 2.4Ghz bands the wn722n is still one of the best option, it is cheap and well supported, unbeatable if you consider the price-quality ratio.