aircrack-ng/rtl8814au #29: Driver update anytime soon?

kmcnabb09, 2021-01-18:
Having loads for problems with the awus1900 and associated rtl8814au. I have tried many different drivers and it “functions” the best with this driver, but I am only able to scan. Can’t capture handshakes, can’t deauth with aireplay or mdk4. Packet Injection doesn’t seem to be working either.
I seem to be stuck on channel -1 when trying to deauth. When put in to monitor mode the interface doesn’t change its name, which I found weird as my other cards change names.

It seems there are a lot of issues associated with this chipset. If there any light at the end of the tunnel or should I return this card and get one more stable that actually works? 
I’d love to see this chipset working as it has great potential and any substitute I can think of getting doesn’t even come close to its range and speeds.

Nj0n, 2021-01-27:
which driver did you installed? i even cant scan... just got on monitor mode and thats it.
i am pissed

kmcnabb09, 2021-01-29:
@Nj0n If you only recently bought a card with the rtl8814au chip I would highly recommend returning it, even if you get it to scan it will
most likely never capture handshakes and such. I replaced mine with a awus036acm and have had no issues with it.

haroldkummer, 2021-01-29:
please update and fix

2chfl, 2021-02-01:
In general, the problem with channel -1 is solved by forcing the desired channel through iwconfig wlan0 channel X (the desired channel). But yes, injections even so, although it works, but with a lot of problems, errors and other things.