aircrack-ng/rtl8814au #30: is it only me or i cant use airodump etc to scan networks?

Nj0n, 2021-01-27:
in fact i did get the monitor mode, but it is not working when trying scanning.
who else got this issue? and how can we slove it ?

AshwinRaam, 2021-01-27:
i am waiting for almost 3 months still no update in injection capabilities. But you can put the device in monitor mode.

Nj0n, 2021-01-27:
> i am waiting for almost 3 months still no update in injection capabilities. But you can put the device in monitor mode.

it is in monitor mode, but i cant do anything with it, even to just connect to wifi i cant.. in fact i can do anything.

haroldkummer, 2021-01-31:
it's incredible. Just trouble with that driver

bertmad3400, 2021-02-03:
> 
> 
> i am waiting for almost 3 months still no update in injection capabilities. But you can put the device in monitor mode.

I have come up with a solution that more or less works for me descriped in #32. No guarrantes, but should hopefully work

patrickps9424, 2021-02-11:
I have the same problem, I can put in monitor mode but in scan never detects any network, I tried (linset) but can't use because doesn't detect any network, any fix for this yet?
Regards

alohl669, 2022-08-28:
> I have the same problem, I can put in monitor mode but in scan never detects any network, I tried (linset) but can't use because doesn't detect any network, any fix for this yet? Regards

Same here, I'm using ubuntu 20.04 with network manager. Airodump can not detect networks. I could get networks info usin nmcli, but airodump does not work :(