aircrack-ng/rtl8814au #32: A very temporary fix for the current problems

bertmad3400, 2021-02-03:
As of lately many people in here have been extremly frustrated with the current state of the RTL8814AU drivers after the update to the 5.10 kernel. I myself have been fighting with the drivers not only for the RTL8814AU but also RTL8812AU chipset over the course of multiple years now, and everytime it seems like a solid solution appears, a new update of linux comes along and breaks it.

Now before actually getting to what I have done to solve this, I just wanted to take a minute to actually give some credit where credit is due. People like @kimocoder often gets a lot of shit for not fixing the drivers when they breaks, but we absolutetly have to remember that without these people we probably wouldn't even have something the resembled working drivers for these incredibly powerful chipsets. Just wanted to let you know that what you do is massivly appreciated, as most of us simply do not have the skillset needed to do what you do.

Now to the solution. What I have discovered is that the **v5.6.4.2** drivers for the rtl8812au chipset works fine for the RTL8814AU chipset too, given some modifications. As these drivers didn't just originally contain the code needed for the support of the RTL8812AU chipset, but also the RTL8821AU and RTL8814AU they do actually work for our use case here. Now I have to warn you, the support for the RTL8814AU was turned OFF in this version of the driver as it got it's own, and therefore this is very far from an officially supported use-case of these drivers, so procced at your own risk, and please **DO NOT** bother the developers with this if it doesn't work. Instead try and do some research yourself first, and then if that doesn't solve it, try and write in this thread, and we can have a look at it together.

Before we begin on the installation process itself, you just need to know a little about some of the quirks of this driver. First of all you should NOT install it with the wireless card plugged in, and you probably want to reboot after installation before pluging it in. You will probably also notice that even after installation, nothing appears when running something like airodump, if you're using virtualbox (might apply to vmware too, havent testet). The fix to this is to make a permanent usb filter for the wirelss card, poweron the machine with the wireless card plugged in using the filter and then after power-on - while the wireless card is plugged in - reboot the guest os. NOT the vm, but the host os. This is done by not powering off the vm, but instead rebooting the guest os by opening a terminal and typing "sudo reboot"

For the more technically inclined amongst you, here's the short version of how to make this work: Clone the **v.5.6.4.2** branch of the aircrack-ng repo for the RTL8812AU drivers, apply [this](https://github.com/aircrack-ng/rtl8812au/files/5862527/0001-Allow-inject-management-and-control-frames.txt) patch, and then undo the changes made to the makefile in [this commit](https://github.com/aircrack-ng/rtl8812au/commit/0f4e46f0fd39f4a18fa01d125284cccaee82ed7b). To last you can either build and install the drivers or use DKMS

This probably didn't make sense to some of you out there, so here's the long explanation of what just happened:

- First off, you need to download a specific branch of the RTL8812AU driver from the aircrack-ng repo. This is done using this command: "_git clone -b v5.6.4.2 https://github.com/aircrack-ng/rtl8812au_"
- Enter the directory ("_cd rtl8812au_" should do the trick)
- Now time for the patch. Download [this](https://github.com/aircrack-ng/rtl8812au/files/5862527/0001-Allow-inject-management-and-control-frames.txt) txt file, put it in the newly downloaded rtl8812au directory, and run the command to apply the patch "_patch -p1 < 0001-Allow-inject-management-and-control-frames.txt_"
- Now we have to modify the makefile in the rtl8812au directory. In your text editor of choice, make the following changes to the makefile:

1. Change line 26 from "_CONFIG_RTL8814A = n_" to "_CONFIG_RTL8814A = y_"
2. Change line 181 from "_ifeq ($(CONFIG_RTL8812A)_$(CONFIG_RTL8821A)_$(CONFIG_RTL8814A), y_y_n)_" to "_ifeq ($(CONFIG_RTL8812A)_$(CONFIG_RTL8821A)_$(CONFIG_RTL8814A), y_y_y)_"
3. Change line 460 from "_#RTL871X = rtl8814a_" to "_RTL871X = rtl8814a_"

- Now we simply need to install the drivers. This can be done either manually or using dkms, though this walktrough will do it manually as I have tested this using dkms. Do however keep in mind that this will require you to build and install the drivers after each kernel update.
- To build the drivers enter the RTL8812AU directory and execute "sudo make" and then "sudo make install". Powroff the machine and follow the steps described earlier regarding usb filters and rebooting the guest os. This should do the trick (hopefully)

Now I hope this resolved the issue for you. If it didn't and you have tried to resolve it yourself, feel free to reply to this, and I will have a look at it. Though please, again, do NOT bother the developers if this doesn't work. This is not a supported way of using these drivers, and the developers are the sole reason we have some more or less working drivers in the first place.

jakenobody, 2021-02-04:
Hello, First, thank for your time, your fix is functional,

airelive, 2021-02-06:
Thanks for sharing this info, bertmad3400, after I tried two days as your way(5.10 kernel or earlier one), still can not use my wifi adapter... I think that's might be my posture's problem, but thank you tough：）

Nj0n, 2021-02-07:
damn !!! i tried, seems it is not working.
i do get monitor mode, but not scanning and no injection

bertmad3400, 2021-02-07:
> 
> 
> damn !!! i tried, seems it is not working.
> i do get monitor mode, but not scanning and no injection
> 
> patching file core/rtw_xmit.c
> Reversed (or previously applied) patch detected! Assume -R? [n]
> you just type "yes" right ?

Hmm, I assumed you remembered the reboot the guest os with the adapter plugged in without closing the whole machine as described right? If you did, I'm afraid there's not a whole lot i can do to help you. I can say, that i did not get a warning about a reversed patch, so my guess would be that you're on the wrong branch? Try and check that the branch is v5.6.4.2.

Nj0n, 2021-02-07:
> Hmm, I assumed you remembered the reboot the guest os with the adapter plugged in without closing the whole machine as described right? If you did, I'm afraid there's not a whole lot i can do to [...]

hmm, yea after installed it , i did a reboot.
then connected the adapter + usb filter - and did sudp reboot.
thats it, right ? 
and if it is important, in my machine i have only usb 1.1

bertmad3400, 2021-02-07:
> hmm, yea after installed it , i did a reboot.
> then connected the adapter + usb filter - and did sudp reboot.
> thats it, right ?
What I think you should have done is to power off the machine completly, then add the usb filter, then boot and the sudo reboot. With that said though, I find it hard to believe that that is the problem.

> and if it is important, in my machine i have only usb 1.1
This hovewer might cause some problems. I assume that your machine does physically have usb 3 port, but VB only allows 1.1 right? Then you should download the extension pack for virtualbox, enable usb 3.0 and try again

Nj0n, 2021-02-07:
> What I think you should have done is to power off the machine completly, then add the usb filter, then boot and the sudo reboot. With that said though, I find it hard to believe that that is the [...]

awesome ! i will try that out and let u know 
and i think i tried also power off the machine after installation , but it is even didnt show me anything that time

skiocink, 2021-02-08:
> damn !!! i tried, seems it is not working.
> i do get monitor mode, but not scanning and no injection
> 
> patching file core/rtw_xmit.c
> Reversed (or previously applied) patch detected! Assume -R? [n]
> you just type "yes" right ?

Same for me. Simply type y then apply the patch again. 

Works fine but the adapter seems to use only 2 antennas

akoposipong, 2021-02-08:
hi guys, im newbie here, 

I believe the instruction is for those using Virtual machine, i was wondering if there is alternative for stand alone kali linux pc

I actually tried the instruction above, however it does not work especially maybe on the reboot part and usb filter since im not using VM.

i recently buy a new alfa awus1900 but having hard time in packet injection.

bertmad3400, 2021-02-08:
> Works fine but the adapter seems to use only 2 antennas
Kinda curious how you found out that it only utilized 2 antennas? Low-level hardware access in linux is far from my strong side

bertmad3400, 2021-02-08:
> hi guys, im newbie here,
>
> I believe the instruction is for those using Virtual machine, i was wondering if there is alternative for stand alone kali linux pc
>
> I actually tried the instruction above, [...]

I, unfortuanetly, do not have a solution for using a installation of kali. As far as I know it's a problem with initializing the drivers, but i don't know how to fix it, and im not even swure that that's the problem

Nj0n, 2021-02-08:
> Same for me. Simply type y then apply the patch again.
>
> Works fine but the adapter seems to use only 2 antennas

what usb do you use?

Nj0n, 2021-02-08:
> What I think you should have done is to power off the machine completly, then add the usb filter, then boot and the sudo reboot. With that said though, I find it hard to believe that that is the [...]

this is working after i swap the usb to usb 3 !!!!! thanks so much !!!

RonaldAM007, 2021-02-17:
Using Kali Linux in WMWare Workstation Player
Alpha AWUS 1900 USB Wireless Adapter

cat /etc/os-release                                                                                          1 ⨯
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2021.1"
VERSION_ID="2021.1"
VERSION_CODENAME="kali-rolling"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"

Following the instructions on patching using the text file.
But getting that FAILED message.
Any ideas or suggestions?

(kali㉿kali)-[~/rtl8812au]
$ sudo patch -p1 < 0001-Allow-inject-management-and-control-frames.txt                                         2 ⨯
patching file core/rtw_xmit.c
Reversed (or previously applied) patch detected!  Assume -R? [n] -R
Apply anyway? [n] y
Hunk #1 FAILED at 4452.
1 out of 1 hunk FAILED -- saving rejects to file core/rtw_xmit.c.rej

sirmick, 2021-02-18:
Works! Tested on the latest Parrot OS, after full upgrade. Used a Alfa AWUS-1900

Skip the patch step - the lines in question in the patch have already been applied. Just modify the Makefile, and follow the instructions for dkms install. Monitor/injection verified.

RonaldAM007, 2021-02-18:
@sirmick Thanks for the response. I will try it and see how it goes.

hellpos, 2021-02-19:
Thanks, very helpful, this workaround worked fine on kali 19.3 version, on 2020 or 2021 I had problems compiling with errors referring to kernel headers.

poweredbycocaine, 2021-03-10:
Using Alfa Awus1900. After initial installation airodump wasn't working, reboot fixed it but injection does not seem to work for me.

Bee1To, 2021-04-24:
Well it is to late for me but don't upgrade to 5.10. I may do a parrot os reinstall with default kernel 5.5 but parrot is a rolling distro so not updating is not an option. I have both rtl8812au and bu, after kernel upgrade to v5.10... NO INJECTION... on both chipset.

4nyp0m0n05, 2021-06-03:
Adapter : Alfa Awus1900
OS : q40s desktop 4.19.0-16-686-pae Debian 4.19.181-1 i686 GNU/Linux
Version : rtl8812au v5.1.5 
git clone -b v4.3.21 https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au
make RTL8814=1
sudo make install RTL8814=1

sudo ip link set wlan0 down
sudo iw dev wlan0 set type monitor
sudo ip link set wlan0 up
sudo iw wlan0 set txpower fixed 3000

It works sniffing mode , aireplay-ng on 2.4GHz for deauthantication frames and mdk3 d for 5GHz

ExpertInLinux, 2021-06-09:
Tried this? https://www.youtube.com/watch?v=0UyI1oMN4VQ

R34v, 2021-07-11:
Hello, can you make a video solving this problem please? I am not good at understanding text. Please

DrShams, 2021-07-16:
yea video will be great cause I stucked too

closed 2023-11-18 by bertmad3400