aircrack-ng/rtl8814au #37: Getting error -16 when trying to enable monitor mode (AWUS 1900)

Colum31, 2021-02-13:
After having installed the driver on my Fedora Laptop, I am getting folowing error:

$ sudo airmon-ng start wlp0s20f0u4

PHY	Interface	Driver		Chipset

phy2	wlp0s20f0u4	8814au		Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac
 (monitor mode enabled)
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wlp0s20f0u4 interface...

Does anyone has an idea how to fix that?

Nj0n, 2021-02-14:
try airmon-ng kill first.
either way try this 
 - 
sudo ifconfig wlp0s20f0u4 down

sudo iwconfig wlp0s20f0u4 mode monitor

sudo ifconfig wlp0s20f0u4 up

terrysimons, 2021-02-25:
iwconfig is the old way of doing things.

Try this:

`sudo iw dev wlp0s20f0u4 set monitor none`

closed 2021-07-25 by Colum31