aircrack-ng/rtl8814au #38: No Injection with RTL8814au AWUS1900

noname1007, 2021-02-17:
I use wifi card running on potassium virtual machine - Virtualbox. Infectious function not working

gazanfarus, 2021-02-19:
I know it's not the best solution, but if you just want to make this thing work, you can downgrade your kernel version. I've been able to make it work on version 4.19 and using older drivers which had both RTL8812au/14au

ArtS, 2021-03-01:
> I know it's not the best solution, but if you just want to make this thing work, you can downgrade your kernel version. I've been able to make it work on version 4.19 and using older drivers which had both RTL8812au/14au

What's your PATCH version? (4.19. <- last number in the semver)

gazanfarus, 2021-03-09:
https://github.com/astsam/rtl8812au/tree/v4.3.21  I used this repo for the driver (note that you have to specify 8814 version)

ExpertInLinux, 2021-06-09:
Have you tried this? https://www.youtube.com/watch?v=0UyI1oMN4VQ