aircrack-ng/rtl8814au #41: Unable to switch to usb 3.0

thindata, 2021-02-20:
When using with an Asus USB-AC68 and applying

`sh -c "echo '1' > /sys/module/8814au/parameters/rtw_switch_usb_mode"`

after a reboot the file reverts to 0

```
cat /sys/module/8814au/parameters/rtw_switch_usb_mode
0
```

```
Device Descriptor:
  bcdUSB               2.00
  idVendor           0x0b05 ASUSTek Computer, Inc.
  idProduct          0x1817 USB-AC68 802.11a/b/g/n/ac (4x4) Wireless Adapter [Realtek RTL8814AU]
```

```
cat /sys/module/8814au/parameters/rtw_switch_usb_mode
0
```

gebau00a, 2021-02-22:
Place a file named 8814au.conf under /etc/modprobe.d 
Put in there
`options 8814au rtw_switch_usb_mode=1`

The option is then applied whenever the module is loaded.

Else, how should the module know which options you want to use?

terrysimons, 2021-02-25:
Can confirm that options 8814au rtw_switch_usb_mode=1 placed in /etc/modprobe.d/8814au.conf works for me:

```
terry@sniffer01:~$ cat /etc/modprobe.d/8814au.conf 
options 8814au rtw_switch_usb_mode=1
terry@sniffer01:~$ cat /sys/module/8814au/parameters/rtw_switch_usb_mode 
1
terry@sniffer01:~$ 
```

thindata, 2021-02-26:
After adding that to the file and rebooting the wifi adapter is no longer detected but if I remove and reinsert it's detected ok and switched to usb 3.0 . What would the cause the issue with it not being detected during boot?

terrysimons, 2021-02-26:
Assuming it's not something like "the interface is down, so it only shows up if you do an ifconfig -a", then that sounds to me like either:

Your port is not a USB 3.0 port. (It should be blue in color).

If it is a USB 3.0 port, then perhaps the kernel is not enabling USB 3 mode for some reason.  Some chipsets require you to specify kernel boot arguments and such... and I've seen XHCI mode be disabled in BIOS or in the kernel before for certain USB chipsets.

I would check your BIOS and your dmesg output and see if you can identify what mode the driver is running in for your USB chipset.

Here's an article talking about forcing USB 2 mode on a USB 3 chipset.  Maybe you can use some tips from this to determine if your chipset is running in USB 2 mode:

https://www.systutorials.com/how-to-force-a-usb-3-0-port-to-work-in-usb-2-0-mode-in-linux/

Here's a relatively old article that talks about getting USB 3.0 working, which might have some insights:

https://askubuntu.com/questions/729558/how-do-i-get-usb-3-0-driver-working-or-check-that-it-is-already-working

terrysimons, 2021-02-26:
Might be a bug in the XHCI drivers on your system, or you had a bad connection, or maybe the cable has issues.

Does it happen every time you boot, or only that one time?

If it happens on every boot, it's probably worth filing a kernel bug and helping them collect info needed to fix the issue.

If it only happens sometimes, you might try replacing your USB 3.0 cable and see if that's any better.

gebau00a, 2021-03-01:
> After adding that to the file and rebooting the wifi adapter is no longer detected but if I remove and reinsert it's detected ok and switched to usb 3.0 . What would the cause the issue with it not being detected during boot?

Check via lsusb -t if the device is shown. 
Also try to switch the mode manually to 0 via the sys interface and check via lsusb again, if it's detected.

I assume the same as Terry, when I switch to USB 3.0 it also falls back to USB 2.0 as my connected cable is too long.