aircrack-ng/rtl8814au #47: make dkms_install fails on Ubunth 20.04

ioannis-vm, 2021-03-03:
(base) johnvm@johnvm-HP-250-G3-Notebook-PC:~/rtl8814au$ sudo make dkms_install
mkdir -p /usr/src/8814au-5.8.5.1
cp -r * /usr/src/8814au-5.8.5.1
dkms add -m 8814au -v 5.8.5.1

Creating symlink /var/lib/dkms/8814au/5.8.5.1/source ->
                 /usr/src/8814au-5.8.5.1

DKMS: add completed.
dkms build -m 8814au -v 5.8.5.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.8.0-44-generic KSRC=/lib/modules/5.8.0-44-generic/build....(bad exit status: 2)
ERROR (dkms apport): binary package for 8814au: 5.8.5.1 not found
Error! Bad return status for module build on kernel: 5.8.0-44-generic (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1/build/make.log for more information.
make: *** [Makefile:2410: dkms_install] Error 10

Why am I getting 'ERROR (dkms apport): binary package for 8814au: 5.8.5.1 not found' ?

gebau00a, 2021-03-12:
I have no Ubuntu installed myself, but are you following the guide for DKMS on Ubuntu as given in the readme?
Seems like you run the normal way, try the Ubuntu special way first

closed 2021-03-20 by ioannis-vm

nyck33, 2022-02-11:
@gebau00a what is the Ubuntu special way?
I get the same error doing this: https://askubuntu.com/a/1302627/806813