aircrack-ng/rtl8814au #48: Former versions to make injection work with AWUS1900?

Lucay00, 2021-03-05:
Hello,

I have just bought the AWUS1900 USB adapter, I am using Kali linux 2021.1 on VMware but I can't succeed in making the injection work with this driver and I do not care to have the latest version of kali linux.
Do you know if it is possible to downgrade linux and get some drivers that can make the AWUS1900 work properly (injection included) and finally take advantage of those four antennas? Which version of kali and the drivers woudl you think would work?

thank you

sally5, 2021-03-08:
What form of install did you go through? Is basic monitor mode packet sniffing working without packet injection?

ArtS, 2021-03-08:
Apparently 4.19 kernel works fine, although I am not sure which patch version. See https://github.com/aircrack-ng/rtl8814au/issues/38

sally5, 2021-03-09:
You can try this driver package though it is a user modified version of the aircrack-ng drivers. I personally have not has success with it but maybe you will.[link](https://github.com/n0ss/realtek-rtl88xxau-dkms)

Lucay00, 2021-03-10:
> You can try this driver package though it is a user modified version of the aircrack-ng drivers. I personally have not has success with it but maybe you will.[link](https://github.com/n0ss/realtek-rtl88xxau-dkms)

Thank you for your answer, it did not worked but I tried this one : https://github.com/morrownr/8814au
And the injection seems to work on the 2021.1 version

sally5, 2021-03-10:
Do you know if this driver works with kali linux?

poweredbycocaine, 2021-03-10:
> 
> 
> Do you know if this driver works with kali linux?

yes it did, but current release of kali seems to be messing with the driver. see #32, as this fixed the issue for me. also if your adapter requires usb 3.0 for its features like packet injection, check the usb port the adapter is attached to, it should be marked blue like [so](https://cdn.shopify.com/s/files/1/0957/6452/files/TIU-3000-1.jpg?4761370832421176751%20).

skiocink, 2021-03-11:
The driver works with Kali upgraded to latest version. I have an AWUS1900.

For Kismet and Bettercap you need to downgrade libpcap to 1.9.1-4 version

For Bettercap is mandatory to put manually the card in monitor mode.

NICE JOB

skiocink, 2021-03-14:
> Thank you for your answer, it did not worked but I tried this one : https://github.com/morrownr/8814au
> And the injection seems to work on the 2021.1 version

I m trying, but I get ACKs 0/0 with aireplay. Both on 2.4 and 5ghz wifi networks

0x7FFFFFFFFFFFFFFF, 2021-07-21:
> I m trying, but I get ACKs 0/0 with aireplay. Both on 2.4 and 5ghz wifi networks

The same here. I'm using AWUS1900.

hostileblue2020, 2021-08-24:
> The same here. I'm using AWUS1900.

Hey man, did you ever figure this out? I just got an AWUS1900 and have the same problem.