aircrack-ng/rtl8814au #50: Build failed under Linux 5.11.6

Zerogoki00, 2021-03-13:
```
DKMS make.log for rtl8814au-r90.27d2344 for kernel 5.11.6-arch1-1 (x86_64)
Sat Mar 13 07:44:21 PM MSK 2021
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.11.6-arch1-1/build M=/var/lib/dkms/rtl8814au/r90.27d2344/build  modules
make[1]: Entering directory '/usr/lib/modules/5.11.6-arch1-1/build'
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl8814au/r90.27d2344/build/hal/hal_intf.o
/var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_ch_switch_notify’:
/var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.c:463:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
  463 |   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8814au/r90.27d2344/build/include/osdep_service_linux.h:94,
                 from /var/lib/dkms/rtl8814au/r90.27d2344/build/include/osdep_service.h:57,
                 from /var/lib/dkms/rtl8814au/r90.27d2344/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7585:6: note: declared here
 7585 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.c:468:5: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  468 |     if (!rtw_cfg80211_allow_ch_switch_notify(adapter))
      |     ^~
/var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.c:470:3: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  470 |   cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl8814au/r90.27d2344/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1800: /var/lib/dkms/rtl8814au/r90.27d2344/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.11.6-arch1-1/build'
make: *** [Makefile:2353: modules] Error 2
```

Arjdroid, 2021-03-18:
There is a pull request and a forked repo that currently works with 5.11.6, check out issue #42

Zerogoki00, 2021-03-19:
Thanks, fix works

closed 2021-03-19 by Zerogoki00