aircrack-ng/rtl8814au #51: Injection working! Check this drivers: https://github.com/morrownr/8814au

kcdtv, 2021-03-14:
Hi there!
Susanita reported in [wifi-libre](https://www.wifi-libre.com/topic-1568-nuevo-driver-para-rtl8814au-awus1900-con-soporte-modo-monitor.html) a successful injection with awus1900 using this drivers: [8814au by morrowr](https://github.com/morrownr/8814au)
I've tried them in Xubuntu LTS (20.04) that uses a kernel 5.4 and injection worked smoothly
Kernel 5.12 (and below) are supported
The drivers are based on v5.8.5.1 (Realtek) (2019-10-29)
It also includes hostapd compatibility
Cheers!

lockywolf, 2021-04-01:
That driver doesn't support setting txpower to anything other than 12db.

kcdtv, 2021-04-01:
The point is to get injection working here, in aircrack-ng driver. Checking morrownr works might help to achieve it. And Txpower is overrated, it is fancy on the box or on advertisement but pretty useless in reality, I'd rather have a device with good sensibility and packet injection than 30dBm. If you can't reach your AP, get a good antenna, It is much better and efficient then to rise txpower that might damage your wifi device on the long term :smiley_cat:

morrownr, 2021-05-18:
> That driver doesn't support setting txpower to anything other than 12db.

It does now.

morrownr