aircrack-ng/rtl8814au #52: Network freeze every few seconds

eric-saintetienne, 2021-03-20:
The network freezes for a few second or two every 10 to 60 seconds, the delay between the freezes seems random.

It's really annoying when editing files remotely as  the keystroke has no effect for a few seconds.

This is what I've tried:

- Try with 5GHz and 2.4GHz networks. Both have random freeze. Example of my 2.4GHz network:
    ```
    wlp0s20f0u4  IEEE 802.11  ESSID:"my-redacted-ssid"  
          Mode:Managed  Frequency:2.437 GHz  Access Point: 01:02:03:04:05:06
          Bit Rate=216.7 Mb/s   Tx-Power=20 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=52/70  Signal level=-58 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
    ```
- plug on other USB ports, same result
- another Wifi USB adapter (TP-LINK TL-WN823N): no freeze
- enabled USB3 using `rtw_switch_usb_mode`: freeze occur in USB2 and USB3

Details:
- Device: TP-Link Archer T9UH v1 [Realtek RTL8814AU] (2357:0106)
- Fedora 33 Workstation (KDE)
- `rtl8814au` driver version: HEAD=27d2344264f774dd2add19d4139dfc07985d6ada
- Network manager:
    ```
    $ nmcli dev show wlp0s20f0u4
    GENERAL.DEVICE:                         wlp0s20f0u4
    GENERAL.TYPE:                           wifi
    GENERAL.HWADDR:                         50:3E:AA:88:A8:76
    GENERAL.MTU:                            1500
    GENERAL.STATE:                          100 (connected)
    GENERAL.CONNECTION:                     my-redacted-ssid
    GENERAL.CON-PATH:                       /org/freedesktop/NetworkManager/ActiveConnection/12
    IP4.ADDRESS[1]:                         192.168.1.2/24
    IP4.GATEWAY:                            192.168.1.1
    IP4.ROUTE[1]:                           dst = 0.0.0.0/0, nh = 192.168.1.1, mt = 600
    IP4.ROUTE[2]:                           dst = 192.168.1.0/24, nh = 0.0.0.0, mt = 600
    IP4.DNS[1]:                             192.168.1.1
    IP4.DOMAIN[1]:                          home
    IP6.ADDRESS[1]:                         fe80::523e:aaff:fe88:a876/64
    IP6.GATEWAY:                            --
    IP6.ROUTE[1]:                           dst = fe80::/64, nh = ::, mt = 256
    ```
- kernel modules:
    ```
    $ lsmod | grep 8814au
    8814au               2383872  0
    cfg80211              954368  4 8814au,rtlwifi,mac80211,rtl8xxxu
    ```

Any idea what causes these sudden network lags? Or how I could get rid of them?

eric-saintetienne, 2021-03-20:
I've attached 5 minutes of pinging the wifi gateway, highlighting when the network freezes: [ping-output-rtl8814au.txt](https://github.com/aircrack-ng/rtl8814au/files/6176314/ping-output-rtl8814au.txt)

I don't want to spoil the fun, but it's usually a 3s to 4s freeze:
```
  1 PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
  2 64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=1.99 ms
  3 64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=1.95 ms
  4 64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=1.96 ms
  5 64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=1.92 ms
  6 64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=2976 ms    <=== HERE
  7 64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=1915 ms    <=== HERE
  8 64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=891 ms     <=== HERE
  9 64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=1.13 ms
 10 64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=1.97 ms
 11 64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=1.94 ms
...
```

During this test, the system was really quiet (99.9% idle!):
```
top - 19:40:05 up 5 days, 11:01,  9 users,  load average: 0.01, 0.07, 0.14
Tasks: 266 total,   1 running, 265 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.0 us,  0.1 sy,  0.0 ni, 99.9 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :  31900.4 total,  12963.8 free,   2632.9 used,  16303.7 buff/cache
MiB Swap:   4096.0 total,   4096.0 free,      0.0 used.  28394.8 avail Mem 
...
```
And I've also attached 5 minutes of the same ping when using another USB Wifi adapter (TP-LINK TL-WN823N, which doesn't exhibit any network lag): [ping-output-using-other-adapter.txt](https://github.com/aircrack-ng/rtl8814au/files/6176315/ping-output-using-other-adapter.txt) with this rtl8192cu (2.4GHz) the latency is better than with the rtl8814au at 5GHz!

eric-saintetienne, 2021-03-20:
Well I feel embarrassed: after a reboot the ping were normal (still high though), all around 2ms, and the lag was gone.
I had an uptime of a a coupld of weeks, working 8h/day work on the machine.

So if it happens to you, it's time to reboot!

closed 2021-03-20 by eric-saintetienne

pmuaddib, 2021-03-25:
I'm having the same issue, reboot didn't help. On W10 same adapter works fine, have no idea why it's happening