aircrack-ng/rtl8814au #53: Packet injection not working for the ALFA AWUS1900

RodrigoCabaco, 2021-03-21:
Hello,
I have installed this driver and the 8812AU drivers for the Alfa Awus1900
It is being detected and I am able to enable monitor mode and make a WiFi network search.
The only problem is that packet injection doesn’t seem to work.
I have tried it on 3 separate machines and I can’t make it work
A quick test on aireplay gives 0/30 on all APs
Kind regards, Rodrigo Cabaco

poweredbycocaine, 2021-03-22:
see #32, this fix problem for many people

haroldkummer, 2021-04-26:
same problem using USB 3 - 5.10.0-kali3-amd64

Bee1To, 2021-04-28:
Ihave the rtl8812au and it stop working after kernel upgrade to v5.10.0  The fix is use rtl8812au v5.6.4.2 with patch. Download from git.
[image]

haroldkummer, 2021-04-28:
> Ihave the rtl8812au and it stop working after kernel upgrade to v5.10.0 The fix is use rtl8812au v5.6.4.2 with patch. Download from git.
> [image]

8814au chipset support is turned off for rtl8812au driver