aircrack-ng/rtl8814au #61: Not Working On kernel 5.12

Arjdroid, 2021-05-07:
I am on Arch Linux x86_64 on a UEFI system with a Netgear A7000 / AC1900 WiFi Card which uses the Realtek RTL8814AU Chipset. 

This repository was not working on Kernel 5.11 so I had to use a fork of the repository which contains the correct patches but now I am on Kernel 5.12 and so the fork has stopped working on this kernel version. But I believe last month there was a merge made to a pull request in order to fix this driver for kernel 5.11...

The fork I was using was made by @gebau00a 

I am not a Kernel / Driver dev so I don't know what changes would be necessary for making the driver work on Kernel 5.12.

I am simply making a thread for others to discuss this issue.

I have also commented below with the compilation errors I'm getting.

Arjdroid, 2021-05-07:
Compilation Error When Running "make" - 

```
# arjdroid at workstation in ~/Downloads/Compiling-Stuff/rtl8814au on git:v5.8.5.1 ● [13:22:56]
→ make

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.12.1-arch1-1/build M=/home/arjdroid/Downloads/Compiling-Stuff/rtl8814au  modules
make[1]: Entering directory '/usr/lib/modules/5.12.1-arch1-1/build'
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_cmd.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_security.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_debug.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_io.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_ioctl_query.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_ioctl_set.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_ieee80211.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_mlme.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_mlme_ext.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_mi.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_wlan_util.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_vht.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_pwrctrl.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_rf.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_chplan.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_recv.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_sta_mgt.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_ap.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/mesh/rtw_mesh.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_xmit.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_p2p.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_rson.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_tdls.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_br_ext.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_iol.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_sreset.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_btcoex_wifionly.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_btcoex.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_beamforming.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_odm.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_rm.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_rm_fsm.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/rtw_rm_util.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/core/efuse/rtw_efuse.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/osdep_service.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/os_intfs.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/usb_intf.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/usb_ops_linux.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/ioctl_linux.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/xmit_linux.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/mlme_linux.o
  CC [M]  /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/recv_linux.o
/home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |    if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                       ^~~~~~~~
/home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/recv_linux.c:358:55: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:271: /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au/os_dep/linux/recv_linux.o] Error 1
make[1]: *** [Makefile:1851: /home/arjdroid/Downloads/Compiling-Stuff/rtl8814au] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.12.1-arch1-1/build'
make: *** [Makefile:2353: modules] Error 2
```

gebau00a, 2021-05-07:
I deleted my repo as it was disheartening to see the pull requests not entertained in 2 months.
I switched to [https://github.com/morrownr/8814au](url) as this works on 5.12, has the LED control features and seems to be quite active.

frostworx, 2021-05-17:
I just received my new Awus1900 and quickly stumbled over this bug after compiling the [AUR package ](https://aur.archlinux.org/packages/rtl8814au-aircrack-dkms-git) failed.
Looks like that `GRO_DROP` being no longer used in 5.12 can be easily fixed though: https://github.com/RinCat/RTL88x2BU-Linux-Driver/commit/657b7cfde9958e273febdeaeac579902e407f577

While writing, I recognized that someone already wrote a fix for this driver:
https://github.com/puuuuh/rtl8814au/commit/d2181e6c1307fefeb28672812e55314e3c7690b3

Would be nice if this went upstream.

frostworx, 2021-05-17:
@gebau00a 
Thanks for the url _(now we only need a PKGBUILD :))_
You mistakenly pointed the url to https://github.com/aircrack-ng/rtl8814au/issues/url though instead of https://github.com/morrownr/8814au

Arjdroid, 2021-05-17:
> @gebau00a
> Thanks for the url _(now we only need a PKGBUILD :))_
> You mistakenly pointed the url to https://github.com/aircrack-ng/rtl8814au/issues/url though instead of https://github.com/morrownr/8814au

I didn't use a PKGBUILD / yay package for the driver, I just installed it as a DKMS module that I compile from source because I don't think there are going to be very many updates to the core driver itself considering the last driver released by realtek was in 2018 and since then all the drivers have only worked on getting the compatibility to work with newer kernels...

They also give you an opportunity to modify configs which is cool like disabling the LED or forcing to go into USB 2.0 mode

frostworx, 2021-05-17:
Sure, but I'd prefer a dkms pkgbuild anyway. Just modifing [this PKGBUILD](https://aur.archlinux.org/packages/rtl8814au-dkms-git) and will ask there if they want to have the changes when done.