aircrack-ng/rtl8814au #67: 8814au-5.8.5.1 does not compile on up-to date Fedora 34

eduvop, 2021-05-25:
TTY output plus output from make is attached. 

Thanks in advance for fixing this,

Eduard

[root@czlx-vopicka rtl8814au]# make dkms_install
mkdir -p /usr/src/8814au-5.8.5.1
cp -r * /usr/src/8814au-5.8.5.1
dkms add -m 8814au -v 5.8.5.1

Creating symlink /var/lib/dkms/8814au/5.8.5.1/source ->
                 /usr/src/8814au-5.8.5.1

DKMS: add completed.
dkms build -m 8814au -v 5.8.5.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=5.12.6-300.fc34.x86_64 KSRC=/lib/modules/5.12.6-300.fc34.x86_64/build......(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.12.6-300.fc34.x86_64 (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1/build/make.log for more information.
make: *** [Makefile:2410: dkms_install] Error 10
[root@czlx-vopicka rtl8814au]# 
[make.log](https://github.com/aircrack-ng/rtl8814au/files/6536767/make.log)

IsaacMvmv, 2021-05-27:
Same line, same file, same error, arch linux

logan001, 2021-06-01:
i guess its k 5.12 and the ways of the linux :)

IsaacMvmv, 2021-06-01:
> i guess its k 5.12 and the ways of the linux :)

Its becouse a bad defined variable I think

iajay, 2021-06-17:
https://github.com/morrownr/8814au - worked for me on fedora 34

flow-control, 2021-07-01:
Patch from https://github.com/aircrack-ng/rtl8814au/pull/64 helped