aircrack-ng/rtl8814au #68: Aireplay isnt working and also the name of the wifi adapter doesnt change after putting monitor mode

BlackstormCoder, 2021-06-06:
Hey guys,
I have Alfa AC1900 wifi adpater and Im using kali 2021.2 with kernal 5.10.0-kali8-amd64 in vmware
I have an issue whenever I put my adpater into monitor mode its name doesnt change to wlan0mon or something. But for scanning the wifi network it works fine with airodump-ng and scan all the devices that are connected to the router.
When its comes to deauth the client it doesnt work.
```
sudo aireplay-ng -0 0 -a ROUTER_MAC -c STATION_MAC wlan0  
04:05:06  Waiting for beacon frame (BSSID: F0:B4:D2:61:30:1B) on channel -1
04:05:06  Couldn't determine current channel for wlan0, you should either force the operation with --ignore-negative-one or apply a kernel patch

```
It shows me this error 
please Help me with this!
Im really desperate to use my new wifi adapter

guofengli854, 2021-06-24:
我的网件A7000也是这种情况，rtl8814AU的芯片组

0x7FFFFFFFFFFFFFFF, 2021-07-21:
> Hey guys,
> I have Alfa AC1900 wifi adpater and Im using kali 2021.2 with kernal 5.10.0-kali8-amd64 in vmware
> I have an issue whenever I put my adpater into monitor mode its name doesnt change to [...]

Have you managed to fix the issue? I have the exactly same problem. I'm using Alfa AWUS1900. I bought it to use in VMware Workstation but it doesn't work. Sigh...

0x7FFFFFFFFFFFFFFF, 2021-07-21:
> 我的网件A7000也是这种情况，rtl8814AU的芯片组

你这个问题解决了吗？我是Alfa AWUS1900的网卡，但是芯片也是rtl8814au。我遇到的问题和楼主描述的几乎是一模一样的。