aircrack-ng/rtl8814au #70: Build fails for Kali 2021 5.4.83-Re4son-v81+

PyMarc2, 2021-08-09:
launches the command `sudo make dkms_install` inside the cloned directory. The log output is this:

```
DKMS make.log for 8814au-5.8.5.1 for kernel 5.4.83-Re4son-v8l+ (aarch64)
Mon Aug  9 16:20:13 UTC 2021
make[1]: Entering directory '/var/lib/dkms/8814au/5.8.5.1/build'
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.4.83-Re4son-v8l+/build M=/var/lib/dkms/8814au/5.8.5.1/build  modules
make[2]: Entering directory '/usr/src/linux-headers-5.4.83-Re4son-v8l+'
Makefile:671: arch/aarch64/Makefile: No such file or directory
make[2]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[2]: Leaving directory '/usr/src/linux-headers-5.4.83-Re4son-v8l+'
make[1]: *** [Makefile:2353: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8814au/5.8.5.1/build'
~
```

Anyonw can help me ?

ozzy80085, 2021-12-25:
im having the same issue and unfortunately i dont have an answer

seijaku-code, 2022-03-01:
use: sudo make dkms_install ARCH=arm64

For some reason the build picks up aarch64 instead (same architecture, but headers don't exist) if you don't specify arch. Happened to me yesterday, built successfully when I defined it.