aircrack-ng/rtl8814au #83: Kali Linux installation fails

Bugatti000, 2021-11-21:
When i try to install the it, it always fails:
make dkms_install
mkdir -p /usr/src/8814au-5.8.5.1
cp -r * /usr/src/8814au-5.8.5.1
dkms add -m 8814au -v 5.8.5.1
Creating symlink /var/lib/dkms/8814au/5.8.5.1/source -> /usr/src/8814au-5.8.5.1
dkms build -m 8814au -v 5.8.5.1

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j16 KVER=5.14.0-kali4-amd64 KSRC=/lib/modules/5.14.0-kali4-amd64/build....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.14.0-kali4-amd64 (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1/build/make.log for more information.
make: *** [Makefile:2410: dkms_install] Error 10

flow-control, 2021-11-29:
Hey there :vulcan_salute:,

what does the `/var/lib/dkms/8814au/5.8.5.1/build/make.log` file look like?
You should see the actual error in there.

Kind regards
Florian

kojibhy, 2021-12-11:
have same issue 
`Error! Bad return status for module build on kernel: 5.13.0-22-generic (x86_64)
Consult /var/lib/dkms/8814au/5.8.5.1/build/make.log for more information.
make: *** [Makefile:2410: dkms_install] Error 10
`

`DKMS make.log for 8814au-5.8.5.1 for kernel 5.13.0-22-generic (x86_64)
Sa 11 Dez 2021 23:59:05 EET
make[1]: Entering directory '/var/lib/dkms/8814au/5.8.5.1/build'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.13.0-22-generic/build M=/var/lib/dkms/8814au/5.8.5.1/build  modules
make[2]: Entering directory '/usr/src/linux-headers-5.13.0-22-generic'
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_vht.o
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.c:3130:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3130 |                 else
      |                 ^~~~
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.c:3133:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3133 |                         goto exit;
      |                         ^~~~
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_br_ext.o
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c: In function ‘validate_mp_recv_frame’:
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c:3575:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3575 |                         for (i = 0; i < precv_frame->u.hdr.len; i = i + 8)
      |                         ^~~
In file included from /var/lib/dkms/8814au/5.8.5.1/build/include/drv_types.h:60,
                 from /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c:17:
/var/lib/dkms/8814au/5.8.5.1/build/include/rtw_debug.h:52:33: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
   52 |         #define RTW_INFO(x,...) do {} while (0)
      |                                 ^~
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c:3578:33: note: in expansion of macro ‘RTW_INFO’
 3578 |                                 RTW_INFO("#############################\n");
      |                                 ^~~~~~~~
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/usb_ops_linux.o
[... 18 lines cut ...]
      |                         ^~~~~~~~~~~~~~
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/ioctl_mp.o
/var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/recv_linux.c:358:76: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |                         if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                                            ^~~~~~~~
[... cut ...]
/var/lib/dkms/8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:474:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  474 |                 cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [Makefile:1879: /var/lib/dkms/8814au/5.8.5.1/build] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.13.0-22-generic'
make[1]: *** [Makefile:2353: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8814au/5.8.5.1/build'
`

Bugatti000, 2021-12-11:
Hey,i already solved the Problem some time ago due to a thread. It was very easy and worked.Bye Gesendet von Mail für Windows Von: Florian EngelhardtGesendet: Montag, 29. November 2021 08:47An: aircrack-ng/rtl8814auCc: Bugatti000; AuthorBetreff: Re: [aircrack-ng/rtl8814au] Kali Linux installation fails (Issue #83) Hey there 🖖,what does the /var/lib/dkms/8814au/5.8.5.1/build/make.log file look like?You should see the actual error in there.Kind regardsFlorian