aircrack-ng/rtl8814au #85: Compile issue on 5.15

ueco-jb, 2021-11-21:
I had basically same compilation error as in issue https://github.com/aircrack-ng/rtl8814au/issues/76.
Patch from [comment](https://github.com/aircrack-ng/rtl8814au/issues/76#issuecomment-939322525) worked on my kernel `5.15.3-arch1-1`.
Is there a reason why this isn't pushed alreday? Or can I just create PR with it?
Br

flow-control, 2021-11-29:
I used the patch from PR #82, works for me

GioGes, 2021-11-30:
i'm new what should i do to make this work on Fedora 35, kernel 5.14.18-300.fc35.x86_64

ueco-jb, 2021-11-30:
How much new are you? Do you know how to use git?
You can check out branch from #82 PR:
```
$ git clone git@github.com:alxgnon/rtl8814au.git
$ cd rtl8814au
$ git checkout fix_5.15
```
and then build and install from there.

GioGes, 2021-11-30:
i've switched from windows 3 weeks ago so i'm pretty bad, i don't really know how to use git but : git checkout fix_5.15 isn't working (error: pathspec 'fix_5.15' did not match any file(s) known to git)

ueco-jb, 2021-11-30:
Did you clone repo I pointed to? Because it's not the `aircrack-ng`, it's some other user's who submitted the fix. Branch is in his repository.
```
$ git clone https://github.com/alxgnon/rtl8814au.git
```

GioGes, 2021-11-30:
ok sorry, did this (Branch 'fix_5.15' set up to track remote branch 'fix_5.15' from 'origin'.
Switched to a new branch 'fix_5.15')
but then make gives me the same error and can't install: 

home/gioges/rtl8814au/os_dep/linux/recv_linux.c:358:76: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:271: /home/gioges/rtl8814au/os_dep/linux/recv_linux.o] Error 1
make[1]: *** [Makefile:1858: /home/gioges/rtl8814au] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.14.18-300.fc35.x86_64'
make: *** [Makefile:2353: modules] Error 2
[gioges@fedora rtl8814au]$ make install
install -p -m 644 8814au.ko  /lib/modules/5.14.18-300.fc35.x86_64/kernel/drivers/net/wireless/
install: cannot stat '8814au.ko': No such file or directory
make: *** [Makefile:2359: install] Error 1

ueco-jb, 2021-11-30:
You can also try this repository, from comments on: https://aur.archlinux.org/packages/rtl8814au-dkms-git/
I got info that it's more up to date with latest kernels.
```
$ git clone https://github.com/morrownr/8814au
```
Also, please read how to use Markdown on GitHub to paste code/logs.

GioGes, 2021-11-30:
thanks a lot for the help, i'll read it

GioGes, 2021-11-30:
This worked and i installed it, after a reboot i can now see my device ed scan for networks but it fails to connect to any network. Do you have any idea?

GioGes, 2021-11-30:
bash{ git clone https://github.com/morrownr/8814au }

ueco-jb, 2021-11-30:
It depends on program which you use to connect to networks. 
Since you're on Fedora, you're probably using `NetworkManager`.
Try reading logs from:
`$ journalctl -u NetworkManager` (CTRL + g to get to the bottom)
Try to get error out of those logs, maybe create a proper question on stackoverflow.com or some Fedora-specific page.
This thread is not a good place to continue such investigation.

GioGes, 2021-11-30:
Thanks again for the help you gave me, i finally made it work, have a nice day