aircrack-ng/rtl8814au #86: Fedora 35 fatal error: net/ipx.h

yuxblank, 2021-11-30:
Hi i'm trying to build dkms on fedora 35 but the build fails:

```sh
DKMS make.log for 8814au-5.8.5.1 for kernel 5.15.5-200.fc35.x86_64 (x86_64)
Wed  1 Dec 00:26:26 CET 2021
make[1]: Entering directory '/var/lib/dkms/8814au/5.8.5.1/build'
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.15.5-200.fc35.x86_64/build M=/var/lib/dkms/8814au/5.8.5.1/build  modules
make[2]: Entering directory '/usr/src/kernels/5.15.5-200.fc35.x86_64'
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.o
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.c:3130:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3130 |                 else
      |                 ^~~~
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_mlme.c:3133:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3133 |                         goto exit;
      |                         ^~~~
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_sta_mgt.o
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c: In function ‘validate_mp_recv_frame’:
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c:3575:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3575 |                         for (i = 0; i < precv_frame->u.hdr.len; i = i + 8)
      |                         ^~~
In file included from /var/lib/dkms/8814au/5.8.5.1/build/include/drv_types.h:60,
                 from /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c:17:
/var/lib/dkms/8814au/5.8.5.1/build/include/rtw_debug.h:52:33: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
   52 |         #define RTW_INFO(x,...) do {} while (0)
      |                                 ^~
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_recv.c:3578:33: note: in expansion of macro ‘RTW_INFO’
 3578 |                                 RTW_INFO("#############################\n");
      |                                 ^~~~~~~~
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_rm_fsm.o
/var/lib/dkms/8814au/5.8.5.1/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[3]: *** [scripts/Makefile.build:277: /var/lib/dkms/8814au/5.8.5.1/build/core/rtw_br_ext.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1872: /var/lib/dkms/8814au/5.8.5.1/build] Error 2
make[2]: Leaving directory '/usr/src/kernels/5.15.5-200.fc35.x86_64'
make[1]: *** [Makefile:2353: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8814au/5.8.5.1/build'
```

running fedora 35,  5.15.5-200.fc35.x86_64

nilfernandezf, 2021-12-01:
Same problem after kernel uupdate 5.15 in pop os

idealwebsolutions, 2021-12-16:
looks like that header was removed in the 5.15 kernel release: https://github.com/strace/strace/issues/201

idealwebsolutions, 2021-12-16:
Saw the same error upgrading to 5.15.6 on Fedora 35. It's due to a missing header in 5.15 kernel; I tried using the latest PR that addresses it but didn't seem to fix the problem.

Recommend using this solution: https://github.com/aircrack-ng/rtl8814au/issues/85#issuecomment-982399647