aircrack-ng/rtl8814au #89: Compile error "/os_dep/linux/recv_linux.c"

brandontict, 2021-12-27:
Running Kali , 
`uname -a
Linux kt0p 5.14.0-kali4-amd64 #1 SMP Debian 5.14.16-1kali1 (2021-11-05) x86_64 GNU/Linux
`
Kali is recently updated . 

Compiling rtl8814au version 5.8.5.1.

**Fatal error while compiling .** 

`/pentest/8814au/rtl8814au/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/pentest/8814au/rtl8814au/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |    if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                       ^~~~~~~~
/pentest/8814au/rtl8814au/os_dep/linux/recv_linux.c:358:55: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [/usr/src/linux-headers-5.14.0-kali4-common/scripts/Makefile.build:276: /pentest/8814au/rtl8814au/os_dep/linux/recv_linux.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.14.0-kali4-common/Makefile:1882: /pentest/8814au/rtl8814au] Error 2
make[1]: *** [/usr/src/linux-headers-5.14.0-kali4-common/Makefile:232: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.14.0-kali4-amd64'
make: *** [Makefile:2353: modules] Error 2
`

zhouxv, 2022-01-18:
kernel 5.13
same problem

averissimo, 2022-01-18:
There are pull requests that fix this issue, I personally use #64, but #82 from @alxgnon might be better for 5.15 compatibility

zhouxv, 2022-01-18:
> There are pull requests that fix this issue, I personally use #64, but #82 from @alxgnon might be better for 5.15 compatibility

-  #89 from @brandontict  works for me.  Ubunut 20.04 LTS / kenel 5.13.0-25-generic
- 3q

brandontict, 2022-01-20:
> There are pull requests that fix this issue, I personally use #64, but #82 from @alxgnon might be better for 5.15 compatibility
Thanks for the feedback. 

I am not too familiar with git, So having issues applying that pull request. 

I've went to the request #82 , clicked on code and copied ' gh pr checkout 82 '. into terminal, under the folder I had cloned for his repo

Also tried

 `git pull origin d2ed7cff6d760dab4b85c819f3ff0b3aca9d2f35 ` 

with no luck. 

I make sure that I am within the folder that I cloned from this repo when attempting to use these pull requests.

alxgnon, 2022-01-20:
@brandontict try this instead:

```
git remote add alxgnon https://github.com/alxgnon/rtl8814au.git
git fetch alxgnon
git checkout --track alxgnon/fix_5.15
```

boeboe, 2022-01-27:
@alxgnon  problem still persists.

    # :~/Downloads$ git clone https://github.com/aircrack-ng/rtl8814au
    Cloning into 'rtl8814au'...
    remote: Enumerating objects: 1121, done.
    remote: Counting objects: 100% (1121/1121), done.
    remote: Compressing objects: 100% (720/720), done.
    remote: Total 1121 (delta 546), reused 944 (delta 385), pack-reused 0
    Receiving objects: 100% (1121/1121), 2.97 MiB | 3.75 MiB/s, done.
    Resolving deltas: 100% (546/546), done.
    # :~/Downloads$ cd rtl8814au/
    # :~/Downloads/rtl8814au$ git remote add alxgnon https://github.com/alxgnon/rtl8814au.git
    # :~/Downloads/rtl8814au$ git fetch alxgnon
    remote: Enumerating objects: 7, done.
    remote: Counting objects: 100% (7/7), done.
    remote: Compressing objects: 100% (1/1), done.
    remote: Total 4 (delta 3), reused 4 (delta 3), pack-reused 0
    Unpacking objects: 100% (4/4), 582 bytes | 194.00 KiB/s, done.
    From https://github.com/alxgnon/rtl8814au
    * [new branch]      fix_5.15   -> alxgnon/fix_5.15
    * [new branch]      main       -> alxgnon/main
    * [new branch]      v5.8.5.1   -> alxgnon/v5.8.5.1
    # :~/Downloads/rtl8814au$ git checkout --track alxgnon/fix_5.15
    Branch 'fix_5.15' set up to track remote branch 'fix_5.15' from 'alxgnon'.
    Switched to a new branch 'fix_5.15'
    # :~/Downloads/rtl8814au$ make
    make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.13.0-1008-intel/build M=/home/boeboe/Downloads/rtl8814au  modules
    make[1]: Entering directory '/usr/src/linux-headers-5.13.0-1008-intel'
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_cmd.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_security.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_debug.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_io.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_ioctl_query.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_ioctl_set.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_ieee80211.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_mlme.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_mlme_ext.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_mi.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_wlan_util.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_vht.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_pwrctrl.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_rf.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_chplan.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_recv.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_sta_mgt.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_ap.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/mesh/rtw_mesh.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/mesh/rtw_mesh_pathtbl.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/mesh/rtw_mesh_hwmp.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_xmit.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_p2p.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_rson.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_tdls.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_br_ext.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_iol.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_sreset.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_btcoex_wifionly.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_btcoex.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_beamforming.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_odm.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_rm.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_rm_fsm.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/rtw_rm_util.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/core/efuse/rtw_efuse.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/osdep_service.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/os_intfs.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/usb_intf.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/usb_ops_linux.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/ioctl_linux.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/xmit_linux.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/mlme_linux.o
      CC [M]  /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.o
    /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
    /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in this function)
      358 |    if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
          |                                                       ^~~~~~~~
    /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.c:358:55: note: each undeclared identifier is reported only once for each function it appears in
    make[2]: *** [scripts/Makefile.build:281: /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.o] Error 1
    make[1]: *** [Makefile:1874: /home/boeboe/Downloads/rtl8814au] Error 2
    make[1]: Leaving directory '/usr/src/linux-headers-5.13.0-1008-intel'
    make: *** [Makefile:2353: modules] Error 2
    # :~/Downloads/rtl8814au$ uname -r
    5.13.0-1008-intel

alxgnon, 2022-01-27:
My patch is for kernel 5.15 and up.

aranthorne, 2022-01-30:
I ran into the compile error also and modified the line 358 of the rtl8814au/os_dep/linux/recv_linux.c to
`if (rtw_napi_gro_receive(&padapter->napi, pskb))`

It compiles successfully for the 5.13.0-27-generic kernel after that change is made.

brandontict, 2022-02-03:
I too am running into the same compile issue, after adding aixgnon commit. 

> @alxgnon problem still persists.
> 

> /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
> /home/boeboe/Downloads/rtl8814au/os_dep/linux/recv_linux.c:358:55: error: ‘GRO_DROP’ undeclared (first use in [...]

`/pentest/alfa/rtl8814au/os_dep/linux/recv_linux.c: In function ‘napi_recv’:
/pentest/alfa/rtl8814au/os_dep/linux/recv_linux.c:358:76: error: ‘GRO_DROP’ undeclared (first use in this function)
  358 |                         if (rtw_napi_gro_receive(&padapter->napi, pskb) != GRO_DROP)
      |                                                                            ^~~~~~~~
/pentest/alfa/rtl8814au/os_dep/linux/recv_linux.c:358:76: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [/usr/src/linux-headers-5.15.0-kali2-common/scripts/Makefile.build:282: /pentest/alfa/rtl8814au/os_dep/linux/recv_linux.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.15.0-kali2-common/Makefile:1892: /pentest/alfa/rtl8814au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-kali2-amd64'
make: *** [Makefile:2353: modules] Error 2
`

I will try and edit line 358 as @aranthorne  stated success with  

Any progress on your end?

u2mejc, 2022-10-26:
There are some other PRs that also fix this, but I happened on this issue / fix first. I had this issue on a fresh install of (K)Ubuntu/22.04, alxgnon's patch plus aranthorne's comment worked for me.

These steps, after fresh install, worked for me:
```
# clone and cd into this repo, then checkout alxgnon's PR
git fetch origin pull/82/head:alxgnon/fix_5.15
git checkout alxgnon/fix_5.15

# Remove undeclared GRO_DROP reference
sed -i'' 's/ \!= GRO_DROP//g' os_dep/linux/recv_linux.c

sudo apt-get update && sudo apt-get upgrade
# Restarted to load kernel changes
sudo apt-get install linux-headers-$(uname -r) make gcc -y
make clean
make
sudo make install
```