aircrack-ng/rtl8814au #95: Errors on 'make' Kali 2022.1

AdamBundschuh, 2022-03-31:
Terminal output: https://pastebin.com/STfMAkbd

This was directly after cloning and cd'ing into the rtl folder.

`┌──(root㉿kali)-[~/rtl8814au]
└─# make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.16.0-kali6-amd64/build M=/root/rtl8814au  modules
make[1]: Entering directory '/usr/src/linux-headers-5.16.0-kali6-amd64'
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali6-common/scripts/pahole-flags.sh: not found
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali6-common/scripts/pahole-flags.sh: not found
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc-11 (Debian 11.2.0-18) 11.2.0
  You are using:           gcc-11 (Debian 11.2.0-19) 11.2.0
/bin/sh: 1: /usr/src/linux-headers-5.16.0-kali6-common/scripts/pahole-flags.sh: not found
  CC [M]  /root/rtl8814au/core/rtw_cmd.o
  CC [M]  /root/rtl8814au/core/rtw_security.o
  CC [M]  /root/rtl8814au/core/rtw_debug.o
  CC [M]  /root/rtl8814au/core/rtw_io.o
  CC [M]  /root/rtl8814au/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8814au/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8814au/core/rtw_ieee80211.o                                                                                                   
  CC [M]  /root/rtl8814au/core/rtw_mlme.o                                                                                                        
/root/rtl8814au/core/rtw_mlme.c: In function ‘rtw_drv_scan_by_self’:
/root/rtl8814au/core/rtw_mlme.c:3130:17: warning: this ‘else’ clause does not guard... [-Wmisleading-indentation]
 3130 |                 else
      |                 ^~~~
/root/rtl8814au/core/rtw_mlme.c:3133:25: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘else’
 3133 |                         goto exit;
      |                         ^~~~
  CC [M]  /root/rtl8814au/core/rtw_mlme_ext.o                                                                                                    
  CC [M]  /root/rtl8814au/core/rtw_mi.o                                                                                                          
  CC [M]  /root/rtl8814au/core/rtw_wlan_util.o                                                                                                   
  CC [M]  /root/rtl8814au/core/rtw_vht.o                                                                                                         
  CC [M]  /root/rtl8814au/core/rtw_pwrctrl.o                                                                                                     
  CC [M]  /root/rtl8814au/core/rtw_rf.o                                                                                                          
  CC [M]  /root/rtl8814au/core/rtw_chplan.o                                                                                                      
  CC [M]  /root/rtl8814au/core/rtw_recv.o                                                                                                        
/root/rtl8814au/core/rtw_recv.c: In function ‘validate_mp_recv_frame’:
/root/rtl8814au/core/rtw_recv.c:3575:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
 3575 |                         for (i = 0; i < precv_frame->u.hdr.len; i = i + 8)
      |                         ^~~
In file included from /root/rtl8814au/include/drv_types.h:60,
                 from /root/rtl8814au/core/rtw_recv.c:17:
/root/rtl8814au/include/rtw_debug.h:52:33: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
   52 |         #define RTW_INFO(x,...) do {} while (0)
      |                                 ^~
/root/rtl8814au/core/rtw_recv.c:3578:33: note: in expansion of macro ‘RTW_INFO’
 3578 |                                 RTW_INFO("#############################\n");
      |                                 ^~~~~~~~
  CC [M]  /root/rtl8814au/core/rtw_sta_mgt.o                                                                                                     
  CC [M]  /root/rtl8814au/core/rtw_ap.o
  CC [M]  /root/rtl8814au/core/mesh/rtw_mesh.o
  CC [M]  /root/rtl8814au/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/rtl8814au/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/rtl8814au/core/rtw_xmit.o
  CC [M]  /root/rtl8814au/core/rtw_p2p.o
  CC [M]  /root/rtl8814au/core/rtw_rson.o
  CC [M]  /root/rtl8814au/core/rtw_tdls.o
  CC [M]  /root/rtl8814au/core/rtw_br_ext.o
/root/rtl8814au/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory
   20 |         #include <net/ipx.h>
      |                  ^~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.16.0-kali6-common/scripts/Makefile.build:292: /root/rtl8814au/core/rtw_br_ext.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.16.0-kali6-common/Makefile:1870: /root/rtl8814au] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.16.0-kali6-amd64'
make: *** [Makefile:2353: modules] Error 2`

burnt43, 2022-04-17:
On Arch I got the same issue. I downgraded
- Linux Kernel to 5.10.9
- Linux Headers to the same
- gcc to 10.2.0
- gcc-libs to the same
- binutils to 2.35.1

realFlowControl, 2022-04-27:
@AdamBundschuh / @burnt43 you can get it to compile and run when applying the patches from #96

suko-mars, 2022-08-13:
The same error on Kali with kernel 5.18

philCryoport, 2022-08-24:
`sudo apt install realtek-rtl8814au-dkms` works for me on 5.18.0-kali5-amd64

fritz1149, 2023-02-14:
you can try this drive: https://github.com/aircrack-ng/rtl8812au
this drive says that it is for "RTL8812AU/21AU and RTL8814AU Wireless drivers"

sagypt, 2023-04-20:
Same issue on

```
lsb_release -a

No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```
```
uname -a

Linux laptop 6.1.0-7-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.20-2 (2023-04-08) x86_64 GNU/Linux
```

FirewallDaProtogen, 2024-05-07:
same issue here on the latest version of linux mint. any ideas?